OFS/opae-sdk #4: about sample for hw？

fengyinzhijian, 2017-11-05:
is there any sample for skx plantform？i want to have a try on skx sdp with systemc.

luebbers, 2017-11-06:
The hello_fpga sample in opae-sdk should work for any platform you have an NLB mode 0 bitstream for - you may have to adjust the AFU_ID in the sample if your NLB's differs so that the sample finds the right AFU.

You can also find a copy of the NLB RTL Verilog source code in the SDK under https://github.com/OPAE/opae-sdk/tree/master/ase/sample_config/intg_xeon_nlb.

If you're using a SystemC workflow to generate AFUs, your best bet is to ask whoever supplied that flow to you for an example. Generating AFU bitstreams is outside of the scope of OPAE (since it's very vendor/tool/platform-specific).

luebbers, 2018-01-05:
I'm closing this for now - please feel free to reopen if you still have questions; or post them on the [OPAE mailing list](http://lists.01.org/mailman/listinfo/opae).

commit 09a0e229 (2023-12-01): fix:CXL traffic generator read/write bandwidth and input arguments bou… (#3054)

* ix:CXL traffic generator read/write bandwidth and input arguments boundary check

fix#1
For the correct value of loop count, the following condition must be satisfied.
max(READ,WRITE) * loop_count <= 268,435,456

fix #2
cannot have READs only (without any previous WRITEs)

fix #3
 READs cannot exceed #WRITEs

Fix #4
Read HDM size from CSR.

* fix code review comments

---------
closed 2018-01-05 by luebbers