OFS/opae-sdk #45: VCS vhdlan -F vs -f and smart_order

TMVector, 2017-11-28:
When using VCS and using VHDL source, vhdlan errors when `-F` is used instead of `-f`. (see [here](https://github.com/OPAE/opae-sdk/blob/master/ase/Makefile#L437))

I think it would also be good to use `-smart_order`.

luebbers, 2017-12-06:
Is this already addressed in a newer patch?

TMVector, 2017-12-08:
Master still has `-F` for vhdlan [here](https://github.com/OPAE/opae-sdk/blob/a74712d9eaef88ca8b414b98410b0bc1f3cd9108/ase/Makefile#L481) (updated line number for latest commit on master)). I cannot see that `-smart_order` has been added anywhere.

I don't know if Questa requires a capital F, but it seems that VCS requires a lowercase f.

michael-adler, 2017-12-11:
It should be fixed now. ModelSim seems to handle relative searches and preprocessor definitions in the VHDL configuration. VCS, strangely, does not.

Thank you for the report.

luebbers, 2017-12-12:
@TMVector can you please let us know if the fix above resolves the issue for you?

TMVector, 2017-12-12:
@luebbers Unfortunately I don't have access to VCS right now, but I will give it a go when I do (might be a few days). Thanks for making the change @michael-adler :)

luebbers, 2018-01-05:
Happy New Year!

@TMVector Did you get a chance to try this again?

TMVector, 2018-01-08:
Happy New Year to you too!

Sadly not. I have a reminder every Monday to look at this again, but it could be another week or two before our licensing issues are sorted out :/

In the mean time feel free to close this issue, and I will re-open it if I have the same problem once I regain access. Thanks again :)

luebbers, 2018-01-23:
Closing for now - please reopen if this is not resolved!

closed 2018-01-23 by luebbers