OFS/opae-sdk #77: OPAE and BSPs

StargazerAlex, 2017-12-06:
Hello OPAE team,

I am eager to try out OPAE but I am a bit confused about how I could possibly get it to run on my Proc10A from Gidel. I am quite new to the FPGA world but I usually use OpenCL together with the Board Support Package from the vendor which provides required drivers and of course the bitstream. I assume that something similar would be required for OPAE but how will that work in the end? Will you provide bitstreams for all platforms? Do the vendors have to build BSPs for OPAE as well? Or is there some magic way to enable me to use the already existing BSPs together with OPAE?

While trying out to run the example code with ASE instead I noticed that the paths between documentation and repository are inconsistent. The documentation (e.g. https://opae.github.io/docs/ase_userguide/ase_userguide.html#running-sample-nlb-in-mcp-configuration) speaks about _<install-prefix>/opae/ase/sample_config/mcp_nlb0/_ which does not exist (anymore).

Thanks a lot in advance!
Best regards,
Alex

luebbers, 2017-12-06:
Hi Alex,

the current OPAE release on GitHub (0.9.0) only supports FPGA cards for Intel Xeon platforms (e.g. the [Programmable Acceleration Card](http://altera.com/pac)) with a matching FIM (that's the bitstream for the static region on the FPGA).

Usually, you would need a FIM implementation (sometimes also called a "blue bitstream") for your FPGA and board that support OPAE. The idea is that the blue bitstream provides enough information to SW via its register map that OPAE can discover, enumerate, and access it. This is not unlike a BSP in concept.  The Intel FPGA platforms supported by OPAE will come with such a bitstream; though in theory, a blue bitstream like that could be implemented for any PCIe-attached FPGA (or even, with some driver modifications, for other interconnects, too). I can't comment on the future availability of blue bitstreams for your particular platform, but I would love to see expanded platform support for OPAE in the future.

Thanks for pointing out the inconsistencies in the documentation - we will address them shortly.

StargazerAlex, 2017-12-07:
Hi luebbers,

thanks a lot for the explanation, it was very clear!

Cheers,
Alex

closed 2017-12-07 by StargazerAlex