OFS/opae-sdk #78: Hello_fpga example

marcomontini, 2017-12-06:
Hi,
I'm using the quick start guide to run hello_fpga but when I launch the command "./hello_fpga", after compiling, I get the following error:
"Error enumerating Accelerators: not found"
Can anyone help me?
Thanks for your help.

luebbers, 2017-12-06:
Hi!

Can you provide some details on your setup? What hardware are you running on, what driver are you using?

Do you have root access to your platform?

Thanks
- Enno

marcomontini, 2017-12-07:
Hi,
I'using a virtual machine with Centos 7.3 (the host OS is Windows server
2012 R2). I installed the OPAE drivers following the guide (v0.9.0). Yes, I
have root access, so I set up my environment variables by root.
Thanks for your help

luebbers, 2017-12-10:
Thanks for the info! What FPGA platform do you have in your system?

marcomontini, 2017-12-10:
I have not a physical platform yet, but I should use an Arria 10 GX FPGA
from Intel/Altera.

Marco Montini

luebbers, 2017-12-12:
If you don't have an FPGA platform in your system, it is expected for the software not to find any accelerators (well, to be fair, there aren't any :)).

If you want to run your OPAE application without hardware, you can run it against a simulation of your accelerator in the AFU Simulation Environment (ASE). You'll need to link your application against a different implementation of the OPAE library (`libopae-c-ase` instead of `libopae-c`) and run your accelerator RTL through a logic simulator.

Have a look at the [ASE user guide](https://opae.github.io/docs/ase_userguide/ase_userguide.html) for more details on how to use ASE.

marcomontini, 2017-12-12:
Yes, I understood the problem after I sent you the email. I followed step
by step the ASE's guide but I had different warnings when I launched the
command "make" (no errors but 37 warnings). Then I launched the command
"make sim" to run the simulator but I got three errors. I attached 3
screens: one after the command "make" and the others after the command
"make install"(it is the same screen divided in two parts). I'm using
licensed Modelsim-ae 10.5c.
Could you help me? Thanks.

Marco Montini
ᐧ

luebbers, 2018-01-05:
Hi Marco,

apologies for the holiday delay; could you please re-share the error messages? I could not see them in your last post.

If you like or it is more convenient, you can also take this discussion to the [OPAE mailing list](http://lists.01.org/mailman/listinfo/opae).

luebbers, 2018-01-05:
Ah, I just saw that you opened another issue with this (Issue #98). I'll close this and reply on the other thread.

closed 2018-01-05 by luebbers

marcomontini, 2018-01-05:
Hi,
yes I sent a message to the following email rahul.r.sharma@intel.com and he
said me to open a new thread. Could you help me? Thanks

Best Regards

Marco Montini

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_6138087535380150665_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>
ᐧ