OFS/opae-sdk #98: OPAE ASE

marcomontini, 2017-12-13:
Hi,
I have several problems with the installation of ASE (I don't have a physical platform). I followed the guide step by step but when I launched the command "make" I got a lot of warnings. Could you help me to understand the problems? I attached a screen of the warnings.
[image]

Thanks for the help

Marco Montini

marcomontini, 2018-01-08:
Hi,
do you have some news about this problem?

Marco Montini

michael-adler, 2018-01-08:
The warnings are coming from a library that ships as part of Quartus.

I'll start with the good news: the code that is triggering the warnings is part of the DDR model for local memory attached to the discrete FPGA. Are you developing for that? If not, the warnings should go away if you set the platform target to it's default: integrated Xeon+FPGA. Furthermore, more recent versions of OPAE SDK use a far simpler (and faster) DDR model that generates fewer warnings.

The bad news: I agree with you that the warnings are ugly. For some reason, the fact that synthesis tools universally spit out so many warnings that they hide actually useful messages appears to be acceptable in EDA circles. This spans all the major vendors. The basic problem is that the set of warnings one user considers insignificant may be significant to someone else. There is a switch within vlog to turn off specific warning numbers, but I'm reticent to set that switch in our scripts in case numbering changes across versions.

marcomontini, 2018-01-08:
I should use a new platform that Intel will launch on market on April. It is called PAC and It is composed by a Xeon processor and an Arria 10 GX FPGA connected by PCIe, so I think the system is not integrated but it will be a discrete system. Despite the warnings I tried to use the "make sim" command (next command in the guide) to run the simulator but I got three errors. I attached a screen of these errors. I used the OPAE SDK 0.9.0 downloaded on https://github.com/OPAE/opae-sdk/releases. Thanks for your help.
[image]
[image]

deepakcu84, 2018-01-08:
Hi, If you want to target ASE for PAC, you need to override the default ASE_PLATFORM during make

$ make ASE_PLATFORM=FPGA_PLATFORM_DISCRETE

I haven't observed the error with ASE_PLATFORM set to DISCRETE. Let me check if I can reproduce this error for the INTG_XEON platform. Will get back to you soon.

michael-adler, 2018-01-08:
Are you using the local memory (DDR) yet? If you are just using the CCI-P signals then you can keep specifying the integrated version until you need local memory. For CCI-P, the only difference between the two is the availability of VL0 and VH1 channels. If you are using eVC_VA then you'll be fine.

If you are using local memory, there have been many changes since 0.9.0, most of them targeting the PAC system. You would probably be better off tracking master for now. On master, you don't need to follow Deepak's advice above. Setting the platform should just work.

marcomontini, 2018-01-09:
I followed the guide using opae-sdk-master downloaded by https://github.com/OPAE/opae-sdk and I'll show the results of the operations. After the installation, I create a new folder 'work' in _/opae-sdk-master/ase_ and then I use the script "create_ase_simbuild_env.sh" (inside 'work' folder) to create an 'ase' folder. Then, inside this new 'ase' folder I use the python script "generate_ase_environment.py". Now, I have several problems with this script. First of all, **what should I pass as argument of this file?** I try to pass the whole 'opae-sdk-master' as argument so it can choose the verilog and vhdl files it needs but in this way I have the error I attach in the screen.  
[image]

michael-adler, 2018-01-09:
Sorry about that. This is an area of active development, specifically for the board you are targeting. The script you referenced is deprecated, but the cmdq0 example has to be fixed. There are updates to examples on branches and I will merge them by the end of the day.

Are you following the ASE User Guide when choosing create_ase_simbuild.sh, or did we miss another document that has to be updated also?

marcomontini, 2018-01-09:
If the pyhton file "generate_ase_environment.py" is deprecated, which script can I use?
Then cmdq0 example is not clear, could you give me more details about this example (tutorial, user guide etc..)?
Yes, I am following the guide avalaible on https://opae.github.io/docs/ase_userguide/ase_userguide.html#ase-examples. What do you mean for "miss another document"? Thanks for your help.

michael-adler, 2018-01-09:
I spent the good part of today updating the user guide to incorporate the many changes over the last few months. I'm not sure when the updated guide will hit the servers.

The basic answer is the recommended script to use is now afu_sim_setup.  create_ase_simbuild.sh is deprecated. generate_ase_environment.py is NOT deprecated, though its use as the script that you should invoke is deprecated.

Much simpler starting examples can be found in the [Basic Build Blocks tutorial](https://github.com/OPAE/intel-fpga-bbb/tree/master/samples/tutorial). The first example will walk you through configuring an ASE environment using the [hw/sim/setup_ase](https://github.com/OPAE/intel-fpga-bbb/blob/master/samples/tutorial/01_hello_world/hw/sim/setup_ase) script. It is a trivial wrapper around afu_sim_setup. The tutorial also walks through compiling and running the simulation -- both RTL and software.

That tutorial builds, by default, in the integrated Xeon mode. To build in discrete mode, pass "--platform=discrete" to setup_ase. You won't see much difference in behavior, though you will see the change in platform configuration in the output of setup_ase (afu_sim_setup).

deepakcu84, 2018-01-10:
Hi Marco,

I was not able to reproduce the first issue using 0.9 release Modelsim 10.5c 64-bit simulator. Can you share details about your environment?
- Modelsim version? 32/64 bit? SE or AE?
Can you try the latest OPAE release? If that doesn't work, try one of the examples from https://github.com/OPAE/intel-fpga-bbb/tree/master/samples/tutorial

Deepak

marcomontini, 2018-01-10:
Hi Deepak,
Yes of course. I am using Modelsim 64 bit AE licensed. I will try to use the example you gave me and I will give you information as soon as possible. Thanks for your help

marcomontini, 2018-01-10:
I have downloaded opae-bbb-master from https://github.com/OPAE/intel-fpga-bbb and I have installed it using the file "install.sh" in path _/intel-fpga-bb-master/samples/tutorial/scripts_. I use the first example "01_hello_world" and type the command _./sim/setup_ase build_ inside _hw_ folder but there is the following error:
_**Error: afu_json_mgr from OPAE SDK failed, parsing /home/intel_tools/intel-fpga-bbb-master/samples/tutorial/01_hello_world/hw/rtl/afu.json**_  (screen attached)
[image]

michael-adler, 2018-01-10:
The error printed by Python indicates that the [jsonschema](https://pypi.python.org/pypi/jsonschema) public package is required but not installed.

marcomontini, 2018-01-10:
Ok I installed the package _jsonschema_ and in this way the error disappears. Continuing to follow the guide with _make_ command I have a new error related to the compiler gcc input files.
[image]

michael-adler, 2018-01-10:
Set the environment variable OPAE_BASEDIR to the root of your OPAE SDK source tree. (The one you cloned from GitHub.)

marcomontini, 2018-01-10:
Previously I've set this environment variable to _opae-sdk-master_ folder. Is it right? Thanks for your advice

michael-adler, 2018-01-10:
In that shell? Doesn't look like it.

<pre>
> ls $OPAE_BASEDIR
AUTHORS          COPYING        TODO    common/   platforms/    scripts/
CMakeLists.txt   README.md      ase/    doc/      safe_string/  tests/
CONTRIBUTING.md  RELEASE_NOTES  cmake/  libopae/  samples/      tools/
</pre>

marcomontini, 2018-01-10:
Ok it is correct. I'll try to go on with next steps. Thanks

marcomontini, 2018-01-12:
I tried the last solution using Basic Build Block tutorial and it works. I have only a warning with the command _make_ before running the simulator. Can it be a problem? The warning is showed in the attached image. Thanks a lot your help.
[image]

deepakcu84, 2018-01-12:
Great! Yes, you may ignore that warning.

closed 2018-01-16 by michael-adler