OFS/opae-sdk #176: fpgaTriggerUmsg() not found in libopae-c-ase.so

eyeloveu, 2018-01-19:
GCC outputs a link error "undefined reference to `fpgaTriggerUmsg'" when I invoke this Umsg function in my host code. 
Finally, I found there is a mismatch between the doc and the c-lib. 
The doc says 'Use fpgaTriggerUmsg() if you need ASE simulation capabilities', while there is no implementation in 'opae-sdk\ase\api\src\umsg.c'. Instead, fpgaGetUmsgPtr() is implemented. 
I don't know which is wrong. Doc error or code bug? 
BTW, is there any reference with more detail about the Umsg APIs?

luebbers, 2018-01-19:
Hi @eyeloveu - thank you for the issue report. It seems like the ASE frontend (libopae-c-ase) does not include an implementation of `fpgaTriggerUmsg()`. We'll add it in a subsequent pull request and update the issue - it'd be great if you could check whether our fix works for you.

Section 3.4 of the [CCI-P specification v0.5](https://01.org/sites/default/files/downloads/opae/intel-fpga-ip-cci-p-inter-spec-external-0.5.pdf) has more details on how UMsgs work.

eyeloveu, 2018-01-21:
@luebbers Thank you!

Can I use fpgaGetUmsgPtr() instead of fpgaTriggerUmsg() to implement the same CPU to FPGA notification function?

luebbers, 2018-01-22:
Yes - that's what `fpgaTriggerUmsg()` does:

    fpgaGetUmsgPtr(handle, &umsg_ptr);
    *((volatile uint64_t *) (umsg_ptr)) = value;

(+ error handling, of course).

Note that this only triggers the **first** UMsg. To trigger others, you'd need to write to different locations of the mapped buffer behind `umsg_ptr`. We're looking at adding that to `fpgaTriggerUmsg()`.

luebbers, 2018-01-23:
@eyeloveu We implemented `fpgaTriggerUmsg()` in PR #182 and merged into master. Can you try if that works for you?

eyeloveu, 2018-01-25:
@luebbers Thanks, It works for me. And I also tried:
*((volatile uint64_t *) (umsg_ptr)+offset) = data_um;
so, I can access the whole cache line, eight uint_64. I don't know whether it represents the value 8 that fpgaGetNumUmsg() returns.  
And can the users leverage multiple cache lines to send more UMsg?
 Is UMAS fixed to 1 CL? Can I expand it?

Other questions:
What's the restriction of fpgaSetUmsgAttributes()? What's the purpose of this function? Only when the least significant bit of the value I write is 1, a UMsg Hint is generated. 

And no matter how I call the UMsg Functions, UMsg ID in the AFU is always 0. Is this a useless field?

luebbers, 2018-01-25:
@eyeloveu Glad that it works for you.

The `fpgaTriggerUmsg()` function in its current form only delivers 64 bits of data with the UMsg (and it can only trigger the first UMsg as noted above). As noted in Section 3.4 of the CCI-P spec, UMsgs can be delivered with 64 *bytes* of data; effectively, the entire cacheline is delivered with the UMsg.

To avoid prefetching effects, each UMsg buffer actually resides in its own page; to be precise, the n-th UMsg uses the n-th cacheline in the n-th page of the UMsg buffer.

    ---------------------
    | UMsg_data #0      | offset 0x0   <- umsg_ptr
    |                   | offset 0x40
    |                   | offset 0x80
           ...
    |                   | offset 0x1000
    | UMsg_data #1      | offset 0x1040
           ...            
    |                   | offset 0x2000
    |                   | offset 0x2040
    | UMsg_data #2      | offset 0x2080

etc.

That's why you always see a `0` in the UMsg ID with your code above - you're always writing to the same cacheline. To trigger UMsg `n`, your code would need to read (off my head, not tested):

    *((volatile uint64_t *)(umsg_ptr)+n*(PAGE_SIZE+8)) = data_um;

the 8 coming from a cacheline having 8*64 bits.

`fpgaSetUmsgAttributes()` defines which of the UMsgs would send a hint on a snoop to the FPGA for ultra-low latency (see Figure 3-2 in the CCI-P spec).

luebbers, 2018-01-25:
If you have any more questions about the usage of OPAE APIs, can I direct you to the OPAE mailing list at https://lists.01.org/mailman/listinfo/opae? Your questions (and the answers) may reach a wider audience there.

I'll close this issue as resolved. Thanks for reporting it!

closed 2018-01-25 by luebbers