OFS/opae-sdk #281: 0.13.0 Driver and Documentation

ZhijieWang, 2018-03-20:
Hi,

Doing some research leads me to this project. Trying to use 0.13.0 release. The documentation is still on 0.9.0, hacked my way to the documentation by padding 0.13.0 to url. 

Could not find Intel Opae Driver sources, seems to be hosted on Intel internal server. Would appreciate an release of that driver either in source file zip or a Debian package.

Thank you,
Bill Wang

tswhison, 2018-04-19:
Please check here for the latest releases/docs: https://github.com/OPAE/opae-sdk/releases.  Alternately, you could manually extract the driver source from the .rpm package.  We'll attach the driver source tarball to the releases page, as well.

luebbers, 2018-04-26:
@mjc0608 , @ZhijieWang : Were you able to get to the driver source code using the path Tim explained above?

ZhijieWang, 2018-04-27:
@luebbers  looking at the release source file and documentation page, it seems the kernel driver is still on Intel's git server.

luebbers, 2018-05-31:
Hi @ZhijieWang - We've included a driver tarball with the OPAE 1.0.0 release (https://github.com/OPAE/opae-sdk/releases/tag/1.0.0-5). It's the file called `opae-intel-fpga-driver-1.0.0.tar.gz`.