OFS/opae-sdk #381: FIM for PAC only supports one PR region in the 1.0 Production Release.

stefanstoeckigt, 2018-04-27:
Currently in the 1.0 Production Release only one PR region is support in the FIM. Allowing you to only configure one AFU per PAC. Will future releases of the FIM support more that one PR region? 

The AFU developer's guide indicated that this should be possible "The FIM includes the static region and one or more PR region partitions for loading AFUs from OPAE.".

tswhison, 2018-06-14:
Thank you for your question.  The current hardware generation will continue to support one PR region per FIM.  Future designs are looking into increasing this number, but I don't have any roadmap that I can share with you at this time.

tswhison, 2018-07-05:
No activity - closing

closed 2018-07-05 by tswhison