OFS/opae-sdk #411: Change OPAE tools CLI

r-rojo, 2018-05-11:
# OPAE Tools Command Line Arguments #

## Introduction ##
Currently, the CLI (command line interface) of the tools included in the OPAE SDK
are not all consistent. In other words, while some tools use `--socket-id` for
identifying the `SocketID` property, others may use `--socket`.

Another example is the use of short options for specifying properties like
bus, device, or function - some tools use `-b, -d, -f` while other use `-B, -D, -F`.

### Common Concepts ###
The following table lists the concepts that have been identified as being
common amongst most tools.

| Concept   | Description                                                            |
|-----------|------------------------------------------------------------------------|
| Bus       | The PCI bus number of an OPAE resource                                 |
| Device    | The PCI device number of an OPAE resource                              |
| Function  | The PCI function number of an OPAE resource                            |
| Socket ID | The socket ID encoded in the BBS                                       |
| Object ID | A 64-bit system-wide unique integer that identifies an OPAE resource   |
| GUID      | A globally unique identifier                                           |
| Timeout   | A maximum amount of time that a specific routine may be allowed to run |
| Help      | Request for help or usage message                                      |

### Proposed Changes ###
After careful review of the different tools and their CLI, the OPAE SW team is planning
on changing the tools identified as being inconsistent in order to have a common set of
command line options for concepts that are common. The following table lists those concepts
along with the long and short command line arguments that will be used for identifying them.

| Concept   | Short Form | Long Form            |
|-----------|------------|----------------------|
| Bus       | `-B`       | `--bus`              |
| Device    | NA         | `--device`           |
| Function  | NA         | `--function`         |
| Socket ID | `-S`       | `--socket-id`        |
| Object ID | NA         | `--object-id`        |
| GUID      | `-G`       | `--guid`             |
| Timeout   |  NA        | `--timeout-<unit> `  |
| Help      |  `-h`      | `--help`             |

_NOTE: Command line options which are less commonly used will not have short options. 
Furthermore, options related to timeouts will also not have short options and will only use long options. They will use the units after in place of `<unit>` in the table listed above
where the untis can be `seconds`, `msec` (for milliseconds), `usec` (for microseconds)._

### OPAE Tools ###
The following sections describe the tools that are part of OPAE SDK and any changes that will be made to bring a unified CLI across all of them.

#### fpgabist ####
`fpgabist` will have its short options changed  to uppercase. Those changes are:
* `-b` will be changed to `-B`
* `-d` will be removed
* `-f` will be removed

#### fpgaconf ####
`fpgaconf` will have its short options changed  to uppercase. Those changes are:
* `-b` will be changed to `-B`
* `-d` will be removed
* `-f` will be removed

#### fpgad ####
`fpgad` does not use any common concepts and will be left unchanged

#### fpgadiag ####
`fpgadiag` will have the following changes applied:
* `-s` will be changed to `-S` (for socket)
* `--bus-number` will be changed to `--bus`
* The option for specifying whether or not the header is displayed will be a long option (`--suppress-hdr`) only since `-S` will be used for socket ID.

#### fpgaflash ####
`fpgaflash` currently supports BDF selection as one argument instead of three. Furthermore, the BDF notation used by `fpgaflash` also supports the inclusion of the PCI domain.
No changes will be made to the CLI of this tool until further notice. 

#### fpgainfo ####
`fpgainfo` will have its short options changed  to uppercase. Those changes are:
* `-b` will be changed to `-B`
* `-d` will be removed
* `-f` will be removed
Socket ID selection may be added in the near future when this tool is ported to C.

#### fpgaport ####
`fpgaport` does not currently share any concepts identified as common. The CLI currently operates on sysfs paths that correspond to OPAE resources.
No changes will be made to the CLI of this tool until further notice.

#### coreidle ####
`coreidle` will have its long option for Socket ID changed from `--socket` to `--socket-id`.

#### hssi_config ####
`hssi_config` will have its long option for Socket ID changed from `--socket` to `--socket-id`.

#### hssi_loopback ####
`hssi_loopback` will have its long option for Socket ID changed from `--socket` to `--socket-id`.

#### packager ####
`packager` does not currently share any concepts identified as common. The CLI currently involves commands or files on the filesystem.
No changes will be made to the CLI of this tool until further notice.

#### usrclk ####
`usrclk` will have its long option for Socket ID changed from `--socket` to `--socket-id`.

#### mmlink ####
`mmlink` will have its long option for Socket ID changed from `--socket` to `--socket-id`.

#### ras ####
`ras` will have its long option for Socket ID changed from `--socket` to `--socket-id`.

asgibbons, 2018-05-11:
We need to reevaluate the PCIe addressing we're using.  PCIe devices are addressed using four fields: the PCIe Segment Group, Bus, Device, and Function.  I believe that if we say we support PCIe, we need to allow for systems with multiple segment groups to get proper addressing.  This could imply an option like "-P 0000:05:00:1" which would be useful in scripts in conjunction with lspci et. al.

I also do not see the need for eschewing short options in favor of long options.  IMHO, this just introduces more potential for error (misspelling option names) and frustration for users.  If we decide on this course, I believe we must add an option to read command-line options from a file, as I would hate to have to type a 200-character command over and over again.

More later...

southerngs, 2018-05-11:
I think we should keep short options for device and function.  Otherwise it is inconsistent if a user wants to specify a full BDF that bus has a short option but the other choices don't.

r-rojo, 2018-05-11:
I have kept the short options for BDF but would prefer to use uppercase letters as opposed to lowercase.
This is in an effort to avoid collisions with other commonly used short options.
I have not included a short option for `object-id`. Should we use one? Should it be `-O` or `-I`?

southerngs, 2018-05-11:
I think your reasoning for using uppercase short options for BDF makes sense.

My vote is for `-O` for `object-id`, but I'd also be okay with long option only for this.

buie1, 2018-06-07:
fpgaconf CLI changed in PR: https://github.com/OPAE/opae-sdk/pull/452

tswhison, 2018-07-05:
Implemented - closing

closed 2018-07-05 by tswhison