OFS/opae-sdk #459: Is there a official in-tree driver available?

mjc0608, 2018-06-07:
In the docs there mentioned a in-tree driver, but the download link is not publicly accessible.

nakulkorde, 2018-06-07:
You can get the driver source code or DEB/RPM package from the release page: https://github.com/OPAE/opae-sdk/releases
The driver source file is named as: opae-intel-fpga-driver-1.0.0.tar.gz

mjc0608, 2018-06-09:
Thanks a lot.
Yes, we can get the great driver. However, it is a module, not a full linux kernel. In the docs it mentioned that the in-tree driver is available in a git repo inside intel. But we cannot access it.

tswhison, 2018-06-21:
Hi @mjc0608, we apologize for any confusion regarding the "in-tree" build.  The documentation will be update to reflect the currently-supported distributions methods for the driver.

tswhison, 2018-07-05:
No activity - closing

closed 2018-07-05 by tswhison