OFS/opae-sdk #566: FME PortFatalErr

stefanstoeckigt, 2018-07-11:
Hi All, 
I am currently developing off the head of the OPAE master branch and getting the following error when i conf my PAC. 

//****** FME ERRORS ******//
Class Path             : /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/errors
Device Path            : /sys/devices/pci0000:3a/0000:3a:00.0/0000:3b:00.0/fpga/intel-fpga-dev.0/intel-fpga-fme.0/errors
Bus                    : 0x3B
Device                 : 0x00
Function               : 0x00
Device Id              : 0x09C4
Catfatal Errors        : 0x00
Errors                 : 0x00
First Error            : 0x00
Next Error             : 0x00
Nonfatal Errors        : 0x08
  PortFatalErr         : 0x01
Pcie0 Errors           : 0x00
Revision               : 0

//****** PORT ERRORS ******//
Class Path             : /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/errors
Device Path            : /sys/devices/pci0000:3a/0000:3a:00.0/0000:3b:00.0/fpga/intel-fpga-dev.0/intel-fpga-port.0/errors
Bus                    : 0x3B
Device                 : 0x00
Function               : 0x00
Device Id              : 0x09C4
Errors                 : 0x10000000000000
First Error            : 0x10000000000000
Revision               : 1

Do you have any information on what is causing this error or what the error means. 
MMIO reads and write work, however events don't seem to be return. 
Any information would be helpful.

buie1, 2018-07-12:
Does this error occur after to attempt a PR of your PAC card? any additional information on what scenario is causing this issue can be helpful.

stefanstoeckigt, 2018-07-13:
The error occurs after I configure the PAC with fpgaconf. I can still run my code on the PAC however events don’t seem to be working on the software side. 
However when I build a design with signaltap the error doesn’t occur after configuring the PAC with fpgaconf and everything seems to work as it should. 
This makes it difficult to track down the cause of the problem. Also I don’t know what the error code means.

tswhison, 2018-07-19:
@buie1 do you guys have CSR spreadsheets for PAC?  I tried decoding the error codes, but the Port error seems to be out of range of the defined bits for MCP.  Thanks

stefanstoeckigt, 2018-07-20:
I also can't seem to clear the error message as root.

[dev@localhost fil_afu]$ sudo fpgainfo errors all -c
//****** FME ERRORS ******//
Class Path             : /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/errors
Device Path            : /sys/devices/pci0000:3a/0000:3a:00.0/0000:3b:00.0/fpga/intel-fpga-dev.0/intel-fpga-fme.0/errors
Bus                    : 0x3B
Device                 : 0x00
Function               : 0x00
Device Id              : 0x09C4
Catfatal Errors        : 0x00
Errors                 : 0x00
First Error            : 0x00
Next Error             : 0x00
Nonfatal Errors        : 0x08
  PortFatalErr         : 0x01
Pcie0 Errors           : 0x00
Revision               : 0

//****** PORT ERRORS ******//
Class Path             : /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/errors
Device Path            : /sys/devices/pci0000:3a/0000:3a:00.0/0000:3b:00.0/fpga/intel-fpga-dev.0/intel-fpga-port.0/errors
Bus                    : 0x3B
Device                 : 0x00
Function               : 0x00
Device Id              : 0x09C4
Errors                 : 0x10000000000000
First Error            : 0x10000000000000
Revision               : 1

ERROR: Could not clear errors for resource /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/errors.
Are you running as root?

tswhison, 2018-07-20:
@r-rojo, could you take a look from the fpgaconf perspective?  Thanks

stefanstoeckigt, 2018-07-27:
This is an one of the .gbs file generated that give the above error. 
[fil_afu.zip](https://github.com/OPAE/opae-sdk/files/2236235/fil_afu.zip)

The problem seems to come and go for repeated builds of the same design. 
All simulations pass and there are no timing error in the build process.

tswhison, 2018-08-30:
Finally got my hands on the CSR spreadsheet.  Error 0x10000000000000 is bit 52, which is "EmifProtocolError - Indicates an EMIF Avalon-mm protocol error."  Hope this sheds some light.

stefanstoeckigt, 2018-09-03:
This helped a lot, thanks. 
The issues has been resolved.

closed 2018-09-03 by stefanstoeckigt