OFS/opae-sdk #579: ASE avs_bfm_inst_ddra incorrect read data valid response number

stefanstoeckigt, 2018-07-18:
The avs_bfm_inst_ddra replied with the incorrect number of readdatavalid (13 instead of 14) when requesting two reads of burst size 8 and 6. 

[image]

tswhison, 2018-07-19:
Hi Tracy, could you have a look/provide next steps?  Thanks

lzhan55, 2018-07-19:
I will look into it.

lzhan55, 2018-07-19:
@stefanstoeckigt Is this a memory read? What does burst side 8 and 6 mean in your test?

stefanstoeckigt, 2018-07-23:
@lzhan55 This is a memory read. With two burst, the first burst count of 8 (512bit Words) and then a burst count of 6 (512bit words). If you look at the readvalid signal it is only valid for 13 clock cycles instead of 14.

lzhan55, 2018-07-24:
Our hardware team will look into it and will get back to you later.

stefanstoeckigt, 2018-07-27:
Attached the .wlf file, navigate to the 101800 time for the bfm memory issue.  
[working_bfm_ddr3_mem_issue.zip](https://github.com/OPAE/opae-sdk/files/2236220/working_bfm_ddr3_mem_issue.zip)

deepakcu84, 2018-08-23:
sorry, was caught up with various release issues. havent' had time to look at this one. I'll have an update next week.

tswhison, 2018-10-25:
Hi Deepak, any update on this one?  Thanks

tswhison, 2019-04-18:
Closing as no traffic.

closed 2019-04-18 by tswhison