OFS/opae-sdk #609: make fail on fedora 28

YouYouExcellent, 2018-08-21:
my gcc version is gcc (GCC) 8.1.1 20180712 (Red Hat 8.1.1-5)
cmake version is 3.11.2

the fail log is:
[ 49%] Building CXX object pyopae/CMakeFiles/_opae.dir/opae.cpp.o
In file included from /home/sse-cd-1/xxx/opae-sdk-master/pyopae/opae.cpp:4:
/home/sse-cd-1/xxx/opae-sdk-master/pyopae/pybind11/include/pybind11/pybind11.h: In member function ‘void pybind11::cpp_function::initialize_generic(pybind11::detail::function_record*, const char*, const std::type_info* const*, pybind11::size_t)’:
/home/sse-cd-1/xxx/opae-sdk-master/pyopae/pybind11/include/pybind11/pybind11.h:317:74: error: cast between incompatible function types from ‘PyObject* (*)(PyObject*, PyObject*, PyObject*)’ {aka ‘_object* (*)(_object*, _object*, _object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Werror=cast-function-type]
             rec->def->ml_meth = reinterpret_cast<PyCFunction>(*dispatcher);
                                                                          ^
cc1plus: all warnings being treated as errors
make[2]: *** [pyopae/CMakeFiles/_opae.dir/build.make:63: pyopae/CMakeFiles/_opae.dir/opae.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:497: pyopae/CMakeFiles/_opae.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

tswhison, 2018-08-30:
Hello, thank  you for pointing this out.  Your issue has been assigned.

YouYouExcellent, 2018-09-07:
it's caused by gcc 8, and this issue will be fixed, if you update pybind11

r-rojo, 2018-10-17:
Sorry for the delay in responding. It seems that updating our version of pybind11 to the latest version will fix these compilation issues. I will update this issue when we update our version and our requirements in setup.py

closed 2018-11-21 by YouYouExcellent