OFS/opae-sdk #610: Support for Broadwell Xeon+FPGA

mjc0608, 2018-08-22:
Does this driver support the Broadwell HARP? Or what should we do to port the new driver to support the old HARP system?

Thanks.

michael-adler, 2018-09-04:
The OPAE driver has been back-ported to Broadwell but the mechanism for building it is messy, to say the least. It isn't really in a form we can cleanly release. We plan to make new Broadwell OPAE machines available soon. Are you asking for a machine you have or for a system in one of the remote access sites?

mjc0608, 2018-09-05:
That should be great. We currently have a machine with SR-5.0.3. By saying you plan to make a new BDX machine with OPAE, do you mean that you will provide a new blue bitstream? Or just reconstruct the whole BDX machine including the hardware? And we also have remote access to the academic vLab, will these machine receive an update one day?

BTW, I also want to ask whether it is possible to release the unclean but usable driver as something like an alpha version? I think we can find a way to build it ourselves. We are developing some vfio-mdev based virtualization technique for the HARP architecture, and OPAE is definitely the best driver for us to work on.

Thanks a lot.

michael-adler, 2018-09-05:
OPAE support on the BDX HARPv2 machines uses the original blue bitstream -- the one that shipped as part of SR-5.0.3. All that has to change is the kernel driver.

While we would like to release the cobbled together BDX driver, the administrative hurdles are simply too large. However, if you put the right OS on your machine (CentOS Linux release 7.5.1804) I can give you a compiled driver. With vLab access you could even copy it yourself in a couple of days. We are running that OS on our FPGA machines.

mjc0608, 2018-09-05:
That should be great also. Can you attach the compiled driver in a email? My email is jcma@umich.edu. Thanks a lot.

closed 2018-09-06 by mjc0608