OFS/opae-sdk #631: Potential problem with packaged /usr/bin/afu_sim_setup in 1.1.0 release

dwilson1394, 2018-09-05:
When using the deb package for ASE 1.1.0 (on a 64-bit Ubuntu 16.04 virtual machine) and attempting to use "afu_sim_setup" in the first tutorial in https://github.com/OPAE/intel-fpga-bbb/tree/master/samples/tutorial/01_hello_world, afu_sim_setup throws an IO exception at line 108 indicating **dst** doesn't exist.

https://github.com/OPAE/opae-sdk/blob/0bf28fa812e9be6e4e6a1a25a229667022d37a76/ase/scripts/afu_sim_setup#L103-L110

Upon further investigation, dst doesn't exist because it is never created via **copy_tree()** from **ase_base** which is set to **/home/lab/sys_aalsdk/opae-sdk-x/ase**. This corresponds to a value returned by getASESrcPath() which uses the local variables **project_src_dir** and **ase_share_dir** . In the package's **/usr/bin/afu_sim_setup**, they appear as:

``` python
def getASESrcPath():
    # CMake will update any variables marked by @ with the proper values.
    project_src_dir = '/home/lab/sys_aalsdk/opae-sdk-x/ase'
    ase_share_dir = '/home/lab/sys_aalsdk/opae-sdk-x/ase'
```

Since ase_share_dir is a non-valid full path on my machine, afu_sim_setup won't work since line 68 (`ase_share = os.path.join(parent_dir, ase_share_dir) `) returns **/home/lab/sys_aalsdk/opae-sdk-x/ase**.  

https://github.com/OPAE/opae-sdk/blob/0bf28fa812e9be6e4e6a1a25a229667022d37a76/ase/scripts/afu_sim_setup#L46-L68

I would guess that ASE_SHARE_DIR variable was originally a relative path and some change in the cmake files has changed it to the packager's full path that results in the os.path.join() in being incorrect.

michael-adler, 2018-09-06:
I'm confused. The user specifies "dst". Is the original complaint that the exception fires because "dst" doesn't exist wrong? Is the reason really that "ase_base" (the source) doesn't exist?

dwilson1394, 2018-09-06:
Sorry about that, I jumped a few steps there.

The IOException with dst is more of an indication of the problem here with the correct ase_share_dir on my machine being /usr/share/opae/ase/ when installing the **opae-ase-1.1.0-2.x86_64.deb** with the command `dpkg -i opae-ase-1.1.0-2.x86_64.deb` and running the command `afu_sim_setup --source hw/rtl/sources.txt build_sim` in the 01_hello_world folder.

The main problem in the deb and rpm packages is the line `ase_share_dir = '/home/lab/sys_aalsdk/opae-sdk-x/ase'` (line 49) for ./usr/bin/afu_sim_setup file. If I'm understanding packaging right, the value here was replaced by cmake to reflect the packager's ase share folder. 

Since ase_share_dir is now a non-valid full path for most users, `ase_share = os.path.join(parent_dir, ase_share_dir) ` (line 68) will now be equal to ase_share_dir since it's a full path. This is what eventually results in copytree not copying the ase_base to dst (since ase_base=ase_share_dir isn't valid for most users) and then then throwing an IOError during rmtree (since dst was not created during copytree).

michael-adler, 2018-09-07:
That's now how it is supposed to work, of course. The way the script was compiled and installed changed while I was on sabbatical. I'll take a look.

michael-adler, 2018-09-07:
Please try the mcadler/ase_share branch and tell me whether your problem is solved.

dwilson1394, 2018-09-08:
From what I can test, the problem is resolved on your branch.

closed 2018-10-09 by michael-adler