OFS/opae-sdk #826: double const in iopll_freq_config

zhouyuan, 2018-10-22:
```
diff --git a/libopae/src/usrclk/user_clk_iopll_freq.h b/libopae/src/usrclk/user_clk_iopll_freq.h
index e64ee2b..8b638d3 100755
--- a/libopae/src/usrclk/user_clk_iopll_freq.h
+++ b/libopae/src/usrclk/user_clk_iopll_freq.h
@@ -58,7 +58,7 @@ struct iopll_config {
 };

 // Reference frequency: 100MHz
-const struct iopll_config const iopll_freq_config[] = {
+const struct iopll_config iopll_freq_config[] = {
        { 0 }, // Freq 0 not configured
        { 0 }, // Freq 1 not configured
        { 0 }, // Freq 2 not configured
```

tswhison, 2018-10-25:
Thank you for pointing this out.  We have also encountered this, and it should be fixed in the next release.

tswhison, 2018-12-13:
Fixed in OPAE 1.2.

closed 2018-12-13 by tswhison