OFS/opae-sdk #1095: OPAE 1.2 Build in Docker Container.

stefanstoeckigt, 2019-01-16:
Hi All, 

I have been trying to get OPAE building in a docker container. However, i am getting the following error at the fitter stage. 

`Info (11888): Total time spent on timing analysis during Post-Routing is 72.62 seconds.
Info (16557): Fitter post-fit operations ending: elapsed time is 00:19:00
Warning (171167): Found invalid Fitter assignments. See the Ignored Assignments panel in the Fitter Compilation Report for more information.
Error (293007): Current module quartus_fit ended unexpectedly. Verify that you have sufficient memory available to compile your design. You can view disk space and physical RAM requirements on the System and Software Requirements page of the Intel FPGA website (http://dl.altera.com/requirements/).
------------------------------------------------
ERROR: Error(s) found while running an executable. See report file(s) for error message(s). Message log indicates which executable was run last.

    while executing
"execute_module -dont_export_assignments -tool fit"
    (procedure "compile_pr_revision_impl" line 17)
    invoked from within
"compile_pr_revision_impl $impl_rev_name"
    (procedure "compile_pr_revision" line 22)
    invoked from within
"compile_pr_revision $options(impl)"
    (procedure "main" line 110)
    invoked from within
"main"
    invoked from within
"if {($::quartus(nameofexecutable) == "quartus") || ($::quartus(nameofexecutable) == "quartus_pro") || ($::quartus(nameofexecutable) == "qpro")} {

        #..."
    (file "./a10_partial_reconfig/flow.tcl" line 1039)
------------------------------------------------
Error (23031): Evaluation of Tcl script ./a10_partial_reconfig/flow.tcl unsuccessful
Error: Quartus Prime Shell was unsuccessful. 2 errors, 3022 warnings
    Error: Peak virtual memory: 729 megabytes
    Error: Processing ended: Fri Jan 11 16:54:55 2019
    Error: Elapsed time: 01:18:29
    Error: Total CPU time (on all processors): 01:19:11
Quartus build failed
makefile:19: recipe for target 'synth_build' failed
make: *** [synth_build] Error 3`

michael-adler, 2019-01-17:
How much memory is assigned to the virtual machine in the container?

stefanstoeckigt, 2019-01-18:
I have run the docker container build on both windows and linux machine with excess of 64GB or ram. The build works when running it on the native system linux OS. However, can't seem to get it working in a container. 
I managed to get it working with the first OPAE release 1.0, however that used Quartus 16.1. 

This is the Dockerfile used to build my container image:
'
FROM docker-registry.garrison.local/os/ubuntu:latest

ARG USER=dev
ARG INSTALLDIR=inteldevstack
ARG INSTALLER=a10_gx_pac_ias_1_2_pv_dev_installer
ENV USER=$USER

USER root
# OPAE build
RUN apt-get update && apt-get install -y \
    build-essential \
    nano \
    curl \
    tar \
    git \
    gcc \
    g++ \
    cmake \
    make \
    autoconf \
    automake \
    libxml2 \
    libxml2-dev \
    libjson-c-dev \
    libboost-all-dev \
    ncurses-dev \
    libncurses-dev \
    libboost-all-dev \
    uuid-dev \
    python \
    python-sphinx \
    python-jsonschema \
    python-git \
    doxygen \
    rsync

RUN apt-get install -y software-properties-common
RUN add-apt-repository -y ppa:deadsnakes/ppa
RUN apt-get update && apt-get install -y \
    python3.6 \
    python3-setuptools \
    python3.6-dev

# Quartus & ModelSim
RUN apt-get install -y \
            ca-certificates \
            lib32ncurses5-dev \
            libcrypto++9v5 \
            libfontconfig1 \
            libglib2.0-0 \
            libsm6 \
            libssl-dev \
            libxrender1 \
            libzmq3-dev \
            expect \
            libtcmalloc-minimal4 \
            lsb libncurses-dev \
            lib32stdc++6 \
            lib32z1 lib32ncurses5

# Intall OPAE
# https://www.altera.com/documentation/dnv1485190478614.html
# Installing the Intel Acceleration Stack Development Package on the Host Machine
USER $USER
WORKDIR /home/$USER

RUN curl http://download.altera.com/akdlm/software/dcp/1.2prv/$INSTALLER.tar.gz | tar xz && \
    sed -i '620i if [ $YESTOALL -eq 1 ] ; then  answer="y"; fi' /home/$USER/$INSTALLER/setup.sh && \
    sed -i '617d' /home/$USER/$INSTALLER/setup.sh && \
    sed -i '54i install_opae=0' /home/$USER/$INSTALLER/setup.sh && \
    /home/$USER/$INSTALLER/setup.sh --yes --installdir /home/$USER/$INSTALLDIR/ && \
    rm -rf /home/$USER/$INSTALLER/
RUN /bin/bash -c "source /home/$USER/inteldevstack/init_env.sh"
ENV QUARTUS_HOME=/home/$USER/$INSTALLDIR/intelFPGA_pro/quartus
ENV PATH="${QUARTUS_HOME}/sopc_builder/bin/:${PATH}"
ENV PATH="${QUARTUS_HOME}/bin:${PATH}"
ENV OPAE_PLATFORM_ROOT=/home/$USER/$INSTALLDIR/a10_gx_pac_ias_1_2_pv
ENV PATH="$OPAE_PLATFORM_ROOT/bin:${PATH}"
ENV LD_PRELOAD=/usr/lib/libtcmalloc_minimal.so.4
#RUN rm /home/$USER/$INSTALLDIR/intelFPGA_pro/quartus/linux64/libboost_system.so
#RUN rm /home/$USER/$INSTALLDIR/intelFPGA_pro/quartus/linux64/libstdc++.so.6

# Install Modelsim
# https://www.altera.com/documentation/uux1498689964626.html
WORKDIR /home/$USER
USER $USER
RUN mkdir /home/$USER/$INSTALLER/
WORKDIR /home/$USER/$INSTALLER/
RUN curl http://download.altera.com/akdlm/software/acdsinst/17.0std/595/ib_installers/ModelSimSetup-17.0.0.595-linux.run > ModelSimSetup-17.0.0.595-linux.run && \
    chmod +x ModelSimSetup-17.0.0.595-linux.run && \
    ./ModelSimSetup-17.0.0.595-linux.run --mode unattended --installdir /home/$USER/$INSTALLDIR/intelFPGA_pro/ && \
    rm -rf /home/$USER/$INSTALLER/
ENV MTI_HOME=/home/$USER/$INSTALLDIR/intelFPGA_pro/modelsim_ase
ENV PATH="${MTI_HOME}/linux/:${MTI_HOME}/bin/:${PATH}"

# Install OPAE-SDK from source
WORKDIR /home/$USER
USER $USER
RUN git clone https://github.com/OPAE/opae-sdk.git /home/$USER/$INSTALLDIR/opae-sdk
RUN mkdir /home/$USER/$INSTALLDIR/opae-sdk/mybuild
RUN cmake -H/home/$USER/$INSTALLDIR/opae-sdk -B/home/$USER/$INSTALLDIR/opae-sdk/mybuild -DBUILD_ASE=1
RUN make -C /home/$USER/$INSTALLDIR/opae-sdk/mybuild
USER root
RUN make install -C /home/$USER/$INSTALLDIR/opae-sdk/mybuild
RUN ldconfig
USER $USER
ENV OPAE_LOC=/home/$USER/$INSTALLDIR/opae-sdk
ENV LD_LIBRARY_PATH="/usr/local/lib64:${LD_LIBRARY_PATH}"

USER $USER
RUN mkdir /home/$USER/development/
WORKDIR /home/$USER/development/
'

michael-adler, 2019-01-18:
Thank you for the detailed configuration.

The use of libtcmalloc_minimal.so makes me nervous. Are you using that only inside the container or also in the build outside the container that works. Have you tried without it?

If not that, I still think it's likely that your container is running out of either RAM or disk space. I don't know docker well. Is it possible that some interaction between the container and the host actually limits the virtual machine's RAM to something lower than the maximum? Perhaps some limit on what can be swapped or pre-allocation of backing storage for anonymous pages in the container?

It's hard to see much that's special in your configuration that would fail during fit (place and route). You got past synthesis and fit didn't complain immediately about a Quartus version mismatch. This tells me your installation is consistent and all the sources and tools were found. It also shows that Quartus can run, update its database, etc. A failure like that in fit is almost always exactly what it says: running out of either memory or disk space.

stefanstoeckigt, 2019-01-18:
Thank you for the quick reply. 

I am only using libtcmalloc_minimal.so in the container build (Not required on the OS Build). 
If i don't set this i get the following error:

```
build	18-Jan-2019 15:59:01	*** Fatal Error: Bus Violation at (nil)
build	18-Jan-2019 15:59:01	Module: quartus_cdb
build	18-Jan-2019 15:59:01	Stack Trace:
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	    0x8453c: cfree + 0x4c (c)
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	     0x65ac: udev_enumerate_scan_devices + 0x26c (udev)
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	   0x278239: Ox3476f8750e1da711 + 0x9 (sys_cpt)
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	   0x249648: Ox3476f8367c6535b1 + 0xc8 (sys_cpt)
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	   0x23a8e7: lc_getid_type + 0x57 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1ee185: cpt_lc_getid_type + 0x47 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1ee1fc: cpt_flexlm_get_local_hostid_by_type + 0x72 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1ee315: cpt_flexlm_get_local_hostid + 0x13 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1ee492: cpt_flexlm_get_config_hostids + 0x17b (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1ee8fb: cpt_flexlm_check_config + 0x159 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1eea31: cpt_flexlm_get_config_from_job + 0xe2 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1eebb9: cpt_flexlm_get_config + 0x125 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1eee7f: cpt_flexlm_get_vendor_string + 0x157 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1e59c2: cpt_ampp_create_license_io + 0x14c (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1e7cee: cpt_ampp_read_license + 0x9c (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1e92b4: cpt_ampp_read_file_header + 0xc84 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1e9c87: cpt_ampp_open_file + 0x52 (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1db6e7: CPT_AMPP_ISTREAMBUF::initialize(std::basic_ifstream<char, std::char_traits<char> >*, char const*, unsigned short, bool) + 0x7f (sys_cpt)
build	18-Jan-2019 15:59:01	   0x1dc333: CPT_AMPP_ISTREAM::open(char const*, int) + 0x1c9 (sys_cpt)
build	18-Jan-2019 15:59:01	    0x238f3: qdb_read_model_header_from_file(QDB_FILE const&, QDB_MODEL_HEADER&) + 0x71 (db_qdb)
build	18-Jan-2019 15:59:01	    0x23bb8: QDB_SERVICE_IMPL::get_model_header_for(QDB_PATH const&) const + 0x120 (db_qdb)
build	18-Jan-2019 15:59:01	    0x24233: QDB_SERVICE_IMPL::get_model_header(QDB_PATH const&) const + 0x3d (db_qdb)
build	18-Jan-2019 15:59:01	    0x1cb44: QDB_SERVICE_IMPL::get_model_headers(QDB_PATH const&, std::list<QDB_MODEL_HEADER*, std::allocator<QDB_MODEL_HEADER*> >&) + 0x5c (db_qdb)
build	18-Jan-2019 15:59:01	    0x1dc30: QDB_SERVICE_IMPL::path_contains_model(QDB_PATH const&) + 0x2a (db_qdb)
build	18-Jan-2019 15:59:01	    0x61f3f: QHD_DATABASE_UTILS::import_ip(std::string const&, std::string const&, std::string const&, std::list<QDB_PATH, std::allocator<QDB_PATH> >&, bool, bool) + 0x145 (comp_qhd)
build	18-Jan-2019 15:59:01	    0x42f35: DSGN_FACADE::import_block(std::string const&, std::string const&, std::string const&, bool) + 0x265 (sys_dsgn)
build	18-Jan-2019 15:59:01	    0x279f1: dsgn_import_block + 0x1b1 (sys_dsgn)
build	18-Jan-2019 15:59:01	    0x4c977: TclNRRunCallbacks + 0x47 (tcl8.6)
build	18-Jan-2019 15:59:01	    0x4e1ca: TclEvalEx + 0x96a (tcl8.6)
build	18-Jan-2019 15:59:01	    0xf68c6: Tcl_FSEvalFileEx + 0x266 (tcl8.6)
build	18-Jan-2019 15:59:01	    0xf69ce: Tcl_EvalFile + 0x2e (tcl8.6)
build	18-Jan-2019 15:59:01	    0x11d8b: qexe_evaluate_tcl_script(std::string const&) + 0x390 (comp_qexe)
build	18-Jan-2019 15:59:01	    0x1611d: qexe_do_tcl(QEXE_FRAMEWORK*, std::string const&, std::string const&, std::list<std::string, std::allocator<std::string> > const&, bool, bool) + 0x6c0 (comp_qexe)
build	18-Jan-2019 15:59:01	    0x1a976: qexe_standard_main(QEXE_FRAMEWORK*, QEXE_OPTION_DEFINITION const**, int, char const**) + 0x297 (comp_qexe)
build	18-Jan-2019 15:59:01	   0x4269cc: qatm_main(int, char const**) + 0x7c (quartus_cdb)
build	18-Jan-2019 15:59:01	    0x3f160: msg_main_thread(void*) + 0x10 (ccl_msg)
build	18-Jan-2019 15:59:01	     0x5b4c: thr_final_wrapper + 0xc (ccl_thr)
build	18-Jan-2019 15:59:01	    0x3f21f: msg_thread_wrapper(void* (*)(void*), void*) + 0x62 (ccl_msg)
build	18-Jan-2019 15:59:01	     0xac5c: mem_thread_wrapper(void* (*)(void*), void*) + 0x5c (ccl_mem)
build	18-Jan-2019 15:59:01	     0x8b49: err_thread_wrapper(void* (*)(void*), void*) + 0x27 (ccl_err)
build	18-Jan-2019 15:59:01	     0x5b8f: thr_thread_wrapper + 0x15 (ccl_thr)
build	18-Jan-2019 15:59:01	    0x41b11: msg_exe_main(int, char const**, int (*)(int, char const**)) + 0xb2 (ccl_msg)
build	18-Jan-2019 15:59:01	   0x423bd6: main + 0x26 (quartus_cdb)
build	18-Jan-2019 15:59:01	    0x20830: __libc_start_main + 0xf0 (c)
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	
build	18-Jan-2019 15:59:01	End-trace
build	18-Jan-2019 15:59:01	
error	18-Jan-2019 15:59:02	------------------------------------------------
error	18-Jan-2019 15:59:02	ERROR: Error(s) found while running an executable. See report file(s) for error message(s). Message log indicates which executable was run last.
error	18-Jan-2019 15:59:02	
error	18-Jan-2019 15:59:02	    while executing
[... 2 lines cut ...]
error	18-Jan-2019 15:59:02	    invoked from within
[... cut ...]
error	18-Jan-2019 15:59:02	    (file "./a10_partial_reconfig/flow.tcl" line 1039)
error	18-Jan-2019 15:59:02	------------------------------------------------
``` 

With regard to the memory usage, i am currently run the container on a system that has 128GB of ram. Docker on linux allow the containers access to the full amount of memory. I have also tried limiting the memory amount allocated to the container with "docker run --memory 64g", same error persists. 

Will check if there are any disk space problems.

michael-adler, 2019-01-18:
I searched for that error on the web. I suspect you have too, since the solution you use is suggested in https://forums.intel.com/s/question/0D50P00003yyTbKSAU/quartus-prime-lite-edition-171-not-running-in-docker-linux.

Another suggestion I found was the --net=host argument to Docker: https://github.com/CentOS/sig-cloud-instance-images/issues/119, which worked for at least one person without changing the allocator.

If that doesn't help, I suggest you ask on the thread at forums.intel.com above. The OPAE environment is layered on top of Quartus. Quartus is built by a different group and nobody working on OPAE can solve this problem.

StaceyRieck, 2021-03-03:
I had this issue too. --net=host is required because the quartus licence check requires a **physical** NIC in order to get a MAC address for the licence check. Without --net=host, all the interfaces are virtual. (As a bonus, when running docker in AWS Codebuild, Add a VPC to the instance to create a physical NIC, otherwise it will also fail this way).