OFS/opae-sdk #1197: software requirement

wangweialkane, 2019-03-11:
Hello OPAE team,

I expected to run OpenCL on a pac_a10 board. I tried to follow the _Intel Acceleration Stack Quick Start Guide for Intel® Programmable Acceleration Card with Intel® Arria®10 GX FPGA_ to install the opae sdk. 
However, I recieve an error when I use:

cmake .. -DBUILD_ASE=1

-- Could NOT find Questa (missing: QUESTA_VSIM_EXECUTABLE QUESTA_VMAP_EXECUTABLE QUESTA_VCOM_EXECUTABLE QUESTA_VLOG_EXECUTABLE QUESTA_VLIB_EXECUTABLE libsvdpi_INCLUDE_DIRS)

I checked all the documents (including the git repository) but do not find any information about the Questa.

I installed the Quartus Prime. Could you tell me how to install the Questa or other ways to solve the problem?

Thanks in advance

buie1, 2019-04-04:
Hey @wangweialkane considering you are trying to run an OpenCL design why are you using the `-DBUILD_ASE` flag?  For running OPAE on a physical device this flag isn't necessary.

when you seeing this error string you are still able to build OPAE, correct?

wangweialkane, 2019-04-13:
Hi @buie1 , in fact, the problem occurred when I tried to install the OPAE rather than run an OpenCL design. The two documents:  [OPAE Installation Guide](https://opae.github.io/0.14.0/docs/install_guide/installation_guide.html) or the [Intel Acceleration Stack Quick Start Guide for Intel® Programmable Acceleration Card with Intel® Arria®10 GX FPGA](https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/ug/ug-qs-ias-v1-1.pdf) show I should use this flag. 

I cannot build the OPAE after the error message.

wangweialkane, 2019-06-28:
Solved. I find this bug caused by my GCC version

closed 2019-06-29 by wangweialkane