OFS/opae-sdk #1307: Unable to locate necessary files for building the simulator

ShashankMG123, 2019-06-10:
I'm unable to locate the file opae-1.3.0.2.x86_64-ase.deb to use the OPAE in the simulation environment.

nakulkorde, 2019-06-13:
Hi @ShashankMG123 , you can find the released installation packages here: https://github.com/OPAE/opae-sdk/releases.

Please close the issue if you are satisfied by the response or feel free to add to it if you need help.

ShashankMG123, 2019-06-14:
@nakulkorde Thanks for the release. On the quick start guide to install OPAE it specifics a file named 'opae-<release>.x86_64-ase.deb' which needs to be installed in order to use OPAE in a simulated environment. Assuming the release to be 1.3.0-2 the file needed would be ' opae-1.3.0-2.x86_64-ase.deb ' which is not found on the release page.

closed 2019-06-25 by ShashankMG123