OFS/opae-sdk #1315: about error information of 'fpgaPrepareBuffer' 's 'invalid parameter'

f1593570, 2019-06-20:
If I use 'fpgaPrepareBuffer' to allocate 1GB's buffer, it can work.
But as I allocate >1GB 's buffer, it shows 'invalid parameter'.
What is the exact meaning of 'invalid parameter'?

tswhison, 2019-08-15:
Have you set up the appropriate number of huge pages?
eg sudo sh -c 'echo 8 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages'

michael-adler, 2019-08-17:
This is the expected behavior. A call to fpgaPrepareBuffer() allocates only a single physical page per call. It rounds up the requested size to whatever page size is required. Requests > 1GB will always fail.

If you need more than 1GB of shared storage you must map them as separate buffers and manage multiple buffer base addresses.

closed 2019-08-22 by tswhison