OFS/opae-sdk #1334: Does fpga-driver support PAC D5005 (S10)?

eyeloveu, 2019-07-29:
I'm currently testing PAC d5005. Yet It seems that the driver does not cooperate with PAC 5005. 
I've tried several versions. However, `fpgainfo fme` always returns `fpgainfo:symbol lookup error: fpgainfo: undefined symbol: strcmp_s` 

So, I doubt whether the driver released on GitHub supports d5005.

There is an rpm release along with DCP2.0 Beta targeting CentOS, but no deb or source code is provided. I tried to dump source code from the rpm package, then compile them by myself, but there are lots of errors. How can I make d5005 and its driver work on Ubuntu?

rweight, 2019-08-15:
If I understand correctly, you are using the RPMs delivered with the dcp 2.0 beta release, extracting the source code and building for Ubuntu?

The strcmp_s error indicates a missing library function. strcmp_s is part of the safe_string library delivered in the opae-libs package.

I don't see anything in your message that specifically indicates a driver related problem. Any driver issues would likely manifest themselves as kernel build problems.

When you ran "fpgainfo fme", had you already built and installed the kernel driver?

eyeloveu, 2019-08-15:
@rweight Yes, I installed the deb driver released from Github before I ran "fpgainfo fme". It seemed that these drivers do not support D5005. Finally, I gave up Ubuntu...

closed 2019-08-29 by tswhison