OFS/opae-sdk #1376: Cannot install or build fpga driver, error: ‘struct erase_info’ has no member named

thierryfischer, 2019-11-15:
Hello,

I am trying to follow the installation guide.
But I am stuck at the very beginning when installing the drivers, it fails with the same error with the .deb package or when building it from sources:
_The machine is a debian 10 (buster)_

```
$ sudo dpkg -i ~/Downloads/opae-intel-fpga-driver-1.3.0-2.x86_64.deb
(Reading database ... 60161 files and directories currently installed.)
Preparing to unpack .../opae-intel-fpga-driver-1.3.0-2.x86_64.deb ...
-e
-e Uninstall of intel-fpga module (version 1.3.0-2) beginning:

------------------------------
Deleting module version: 1.3.0-2
completely from the DKMS tree.
------------------------------
Done.
-e Force regeneration of new initramfs
/var/lib/dpkg/info/opae-intel-fpga-driver.prerm: 16: /var/lib/dpkg/info/opae-intel-fpga-driver.prerm: dracut: not found
Unpacking opae-intel-fpga-driver (1.3.0-2) over (1.3.0-2) ...
Setting up opae-intel-fpga-driver (1.3.0-2) ...
Add module source to dkms

Creating symlink /var/lib/dkms/intel-fpga/1.3.0-2/source ->
                 /usr/src/intel-fpga-1.3.0-2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j48 KERNELRELEASE=4.19.0-6-amd64 KERNELDIR=/lib/modules/4.19.0-6-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.0-6-amd64 (x86_64)
Consult /var/lib/dkms/intel-fpga/1.3.0-2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j48 KERNELRELEASE=4.19.0-6-amd64 KERNELDIR=/lib/modules/4.19.0-6-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.0-6-amd64 (x86_64)
Consult /var/lib/dkms/intel-fpga/1.3.0-2/build/make.log for more information.
$ cat /var/lib/dkms/intel-fpga/1.3.0-2/build/make.log
DKMS make.log for intel-fpga-1.3.0-2 for kernel 4.19.0-6-amd64 (x86_64)
Fri 15 Nov 2019 02:26:31 PM CET
make -C /lib/modules/4.19.0-6-amd64/build M=/var/lib/dkms/intel-fpga/1.3.0-2/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-6-amd64'
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/altera-asmip2.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/misc/avmmi-bmc.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/fpga-mgr.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/uuid_mod.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/pcie.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/pcie_check.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/feature-dev.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/fme-pr.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/fme-iperf.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/fme-dperf.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/fme-error.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/fme-main.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/backport.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/afu.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/region.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/dma-region.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/afu-error.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/afu-check.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/intel-fpga-pac-hssi.o
  CC [M]  /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/fpga/intel/intel-fpga-pac-iopll.o
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c: In function ‘mtd_erase_callback’:
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:55:11: error: ‘struct erase_info’ has no member named ‘mtd’
  if (instr->mtd->_erase == part_erase) {
           ^~
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:56:37: error: ‘struct erase_info’ has no member named ‘mtd’
   struct mtd_part *part = PART(instr->mtd);
                                     ^~
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:36:39: note: in definition of macro ‘PART’
 #define PART(x)  ((struct mtd_part *)(x))
                                       ^
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:62:11: error: ‘struct erase_info’ has no member named ‘callback’
  if (instr->callback)
           ^~
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:63:8: error: ‘struct erase_info’ has no member named ‘callback’
   instr->callback(instr);
        ^~
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c: In function ‘spi_nor_erase’:
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:589:7: error: ‘struct erase_info’ has no member named ‘state’
  instr->state = ret ? MTD_ERASE_FAILED : MTD_ERASE_DONE;
       ^~
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:589:23: error: ‘MTD_ERASE_FAILED’ undeclared (first use in this function); did you mean ‘MTD_WRITEABLE’?
  instr->state = ret ? MTD_ERASE_FAILED : MTD_ERASE_DONE;
                       ^~~~~~~~~~~~~~~~
                       MTD_WRITEABLE
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:589:23: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.c:589:42: error: ‘MTD_ERASE_DONE’ undeclared (first use in this function); did you mean ‘DL_STATE_NONE’?
  instr->state = ret ? MTD_ERASE_FAILED : MTD_ERASE_DONE;
                                          ^~~~~~~~~~~~~~
                                          DL_STATE_NONE
make[4]: *** [/usr/src/linux-headers-4.19.0-6-common/scripts/Makefile.build:309: /var/lib/dkms/intel-fpga/1.3.0-2/build/drivers/mtd/spi-nor/spi-nor.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-4.19.0-6-common/Makefile:1534: _module_/var/lib/dkms/intel-fpga/1.3.0-2/build] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-6-amd64'
make: *** [Makefile:53: all] Error 2
```
Does anyone knows how to fix this ?

Best regards,

tswhison, 2019-12-12:
Hello, we are sorry that you are running into trouble building the drivers.  Unfortunately, for release 1.3.0, only kernel versions 3.10 and 4.7 are officially tested/supported.  Support for more recent kernel versions requires some amount of porting effort.

rweight, 2019-12-12:
The spi-nor source was taken directly from the 4.12 kernel. It is being carried in the installation package because the spi-nor module is not included in the 3.10 kernel. Try removing this module from your build - it should already be there in the 4.19 kernel.

thierryfischer, 2020-01-06:
Hi, I did not try to remove this module from build, but I saw that the new release  https://github.com/OPAE/opae-sdk/releases/tag/1.4.0-1 adds support to the "community 4.19 LTS kernels" hence with this release there is no issue installing the driver package.
Thank you @tswhison & @rweight for your help.

closed 2020-01-06 by thierryfischer