OFS/opae-sdk #1410: Debian packages for 1.4.0

sampsyo, 2020-01-06:
Hello! I'm trying to follow the [installation guide](https://opae.github.io/latest/docs/install_guide/installation_guide.html) on an Ubuntu machine. The guide mentions that there should be `.deb` packages available to download, but [the 1.4.0 release](https://github.com/OPAE/opae-sdk/releases/tag/1.4.0-1) doesn't have any `.deb` files under Assets. I notice that versions 1.3.0 and earlier do—could you please generate and attach Debian packages as well? Thank you!

nakulkorde, 2020-01-08:
Hi @sampsyo , for 1.4.0 release, Ubuntu is not a supported OS and hence we did not go through the validation cycle for Ubuntu. You can still generate the packages from source by following the [documentation](https://opae.github.io/1.4.0/docs/install_guide/installation_guide.html).

sampsyo, 2020-01-08:
OK, thanks!!

closed 2021-10-11 by tswhison