OFS/opae-sdk #1417: Intel Quartus Pro License

stefanstoeckigt, 2020-01-14:
As of 01/01/2020 the following have become Unlicensed with the [Acceleration Stack Version 1.2](https://www.intel.com/content/www/us/en/programmable/products/boards_and_kits/dev-kits/altera/acceleration-card-arria-10-gx/getting-started.html).

```
+-----------------------------------------------------------------------------+
; Assembler Encrypted IP Cores Summary                                        ;
+--------+-----------------------------------------------------+--------------+
; Vendor ; IP Core Name                                        ; License Type ;
+--------+-----------------------------------------------------+--------------+
; Altera ; PCIe SRIOV with 4-PFs and 2K-VFs (6AF7 00FB)        ; Unlicensed   ;
; Altera ; Low Latency 10Gbps Ethernet MAC (6AF7 0119)         ; Unlicensed   ;
; Altera ; Low Latency 40Gbps Ethernet MAC and PHY (6AF7 011B) ; Unlicensed   ;
; Altera ; Signal Tap (6AF7 BCE1)                              ; Licensed     ;
; Altera ; Signal Tap (6AF7 BCEC)                              ; Licensed     ;
+--------+-----------------------------------------------------+--------------+
```

```
4315 Info (16677): Loading final database

4316 Warning (292015): License for core 6AF7_0119, version 0000.00 is expired.

4317 Warning (292015): License for core 6AF7_00FB, version 0000.00 is expired.

4318 Warning (292015): License for core 6AF7_011B, version 0000.00 is expired.
```

Please advise to what is required to get a new license?

cschlaak, 2020-01-19:
I suddenly have the exact same problem, which also seems to break the synthesis. Also, someone else in the intel forum is fighting this issue: https://forums.intel.com/s/question/0D50P00004ZMIyk/error-213009-file-name-outputfilesafufitgreenregionpmsf-does-not-exist-or-cant-be-read?language=en_US

stefanstoeckigt, 2020-01-29:
License Patch for Intel Acceleration Stack v1.2 for Intel FPGA PAC with Intel Arria 10 GX:

[Link](https://www.intel.com/content/www/us/en/search.html?ws=text#q=acceleration%20stack%20patch&t=All)

[Download](https://cdrdv2.intel.com/v1/dl/getContent/618887?wapkw=acceleration%20stack%20patch)

closed 2020-01-29 by stefanstoeckigt