OFS/opae-sdk #1453: Shared FPGA access unsupported by upstream dfl-afu driver

ddcc, 2020-02-12:
Is shared FPGA access supported by the upstream driver? With the old `intel-fpga-afu` driver, I see that `afu_open` supports both exclusive and non-exclusive access (`feature_dev_use_begin`/`feature_dev_use_excl_begin`). But with the new `dfl-afu` driver, `dfl_feature_dev_use_begin` does a test-and-set on a per-device lock.

tswhison, 2020-02-14:
Hello, shared FPGA access support is WIP on upstreaming.  See https://lkml.org/lkml/2019/12/30/34

ddcc, 2020-02-14:
Thanks for the link, nice to know that it's in development. I see that there hasn't been much activity since the original patch was posted, is it just waiting to be picked into the linux-next tree?

closed 2020-06-15 by ddcc