OFS/opae-sdk #1461: Poorly named driver source release file

jsomervi, 2020-02-18:
In the latest release we have:

opae-intel-fpga-driver-source-2.0.1-6.x86_64.rpm

which appears as a binary x86_64 rpm by naming convention.  Please rename it to opae-intel-fpga-driver-2.0.1-6.src.rpm or opae-intel-fpga-driver-x86_64-2.0.1-6.src.rpm if you insist on having x86_64 in the name.  Even better yet, if you're not including a spec file then just release the source as a compressed tarball.  As it stands, this breaks some of our tools/scriptware.   Thanks.

nakulkorde, 2020-02-19:
Following are the instructions for generating binary RPM from the ```opae-intel-fpga-driver-source-a.b.c-d.x86_64.rpm``` that we provided with the release.

Download the kernel source package from shareholders. We assume the kernel package name is: 
```opae-intel-fpga-driver-source-a.b.c-d.x86_64.rpm ```
 
Before we install the latest rpm driver, it had better remove the installed driver firstly. 
```# rpm –e opae-intel-fpga-driver ```
 
Install the new OPAE source rpm. 
```# yum install opae-intel-fpga-driver-source-a.b.c-d.x86_64.rpm ```
 
Installing the kernel driver .rpm will cause the driver source (packaged inside the .rpm) to be placed in the /usr/src directory. Change directory there and follow these steps: 
 
```# cd /usr/src 
# tar zxf opae-intel-fpga-driver-source-a.b.c-d.tar.gz 
# rpmbuild –tb opae-intel-fpga-driver-source-a.b.c-d.tar.gz ```
 
This will create the installable binary rpm in ~/rpmbuild/RPMS/x86_64. 
 
Change directory to there and install the binary rpm. 
```# cd ~/rpmbuild/RPMS/x86_64 
# yum install opae-intel-fpga-driver-a.b.c-d.rpm ```

jsomervi, 2020-02-19:
That's not what I asked.  The file opae-intel-fpga-driver-source-a.b.c-d.x86_64.rpm is poorly named, it looks like it is a binary rpm by naming convention, but it contains source.  Please re-read what I wrote previously.

nakulkorde, 2020-02-20:
Sorry about that. I have uploaded the source tarball here:
https://github.com/OPAE/opae-sdk/releases/download/1.3.6-4/opae-intel-fpga-driver-2.0.1.tar.gz

Hope this helps.

jsomervi, 2020-02-20:
That's great, thanks.  Now can you do the same for the 2.0.4 driver release please (in the 1.4.0 SDK release) and any releases going forward from now on?  Thanks!

GrueMaster, 2020-05-20:
Actually, this is named correctly, for the most part.  To truly be accurate, it should be named opae-intel-fpga-driver-2.0.1-6.noarch.rpm and there should be a corresponding opae-intel-fpga-driver-2.0.1-6.src.rpm.  The <arch>.rpm is a package installation rpm containing preinstall, postinstall, and postrm scripts as well as the driver source, which is needed to rebuild the drivers outside of the kernel during a kernel update (dkms handles this).  The src.rpm is normally a special rpm package that has the source tarball, teh rpm spec file, and possibly any patches to the original source.

Since I doubt Intel will be supporting these drivers on any other architecture anytime soon (PPC, ARM64, etc), x86_64 is appropriate in this instance, even though technically it could be noarch as there are no real binaries in the package.  Now, if the drivers are using kernel functions that are specific to x86_64, then the x86_64.rpm is in fact the correct naming convention.