OFS/opae-sdk #1482: Support for 256-bit MMIO writes

ddcc, 2020-03-04:
Are 256-bit AVX MMIO writes supported? I see that they're not currently supported by the ASE emulator, and I can write a patch to support this, but I'm not sure if they are supported by the underlying blue bitstream? When I look at the definition of the `t_ccip_c0_ReqMmioHdr` struct, only three fixed values are defined for the 2-bit `length` field:

```
typedef struct packed {
    t_ccip_mmioAddr address;    // [27:12]  4B aligned Mmio address
    logic [1:0]     length;     // [11:10]  2'b00- 4B, 2'b01- 8B, 2'b10- 64B
    logic           rsvd;       // [9]      reserved, don't care
    t_ccip_tid      tid;        // [8:0]
} t_ccip_c0_ReqMmioHdr;
```

tswhison, 2020-03-05:
Thank you for your query.  Here is a response from our hardware team regarding the blue bits:

> MMIO only supports 32 to 64 bit writes. There are currently no plans to support this in the near future.

ddcc, 2020-03-05:
Just to clarify, aren't 512-bit MMIO writes supported as well? I see that software support has landed recently in OPAE.

tswhison, 2020-03-05:
Here's another reply from the hardware team:

> There’s an encoding for 512 bit MMIO, too. CCI-P on the HW side doesn’t define an encoding for 256 bits. I’m not sure why – especially since it was first defined for BDX, which doesn’t easily move 512 bits in a single instruction.

> It will be easier to support it on OFS, since PCIe TLPs will reach all the way to the AFU. We could revisit it then.

ddcc, 2020-03-05:
Sorry, what is OFS? I see an ofs-platform-afu-bbb repository, is that a newer internal framework for building FPGA platforms?

nathanborak, 2024-10-14:
Closing due to inactivity/age. If you still require assistance, please re-open the issue.