OFS/opae-sdk #1537: Support for write-combining

ddcc, 2020-04-22:
Given that 64-byte PCIe writes are supported (#1482), it would be good to have official support for write-combining memory-mapped regions in both the kernel driver and OPAE SDK to achieve these without native AVX-512 support.

For now, I am manually modifying the kernel driver to call `pgprot_writecombine` instead of `pgprot_noncached`.

See e.g. https://www.intel.com/content/dam/www/public/us/en/documents/white-papers/pcie-burst-transfer-paper.pdf

tswhison, 2020-04-23:
Hello, actually 64-byte MMIO writes are not supported.  I think you may have been referring to this 'MMIO only supports 32 to 64 bit writes' from #1482.

ddcc, 2020-04-23:
There was a second response that stated `There’s an encoding for 512 bit MMIO, too`, and the CCI-P specification shows a 64-byte encoding: ` 2'b00- 4B, 2'b01- 8B, 2'b10- 64B`. The OPAE libraries also have `fpgaWriteMMIO512` and `write_csr512` functions that I've used with my design in simulation.

tswhison, 2020-04-23:
Yes, you are correct in that simulation supports them by virtue of those APIs.  However, the actual hardware doesn't support the 512-bit MMIO, and the statement from the hardware team regarding this is 
> MMIO only supports 32 to 64 bit writes. There are currently no plans to support this in the near future.

ddcc, 2020-04-23:
I see, the current situation is a bit confusing. But it does explain why I haven't been able to get larger writes working with hardware.

closed 2021-10-11 by tswhison