OFS/opae-sdk #1561: Support for HSSI config and testing

Hecmay, 2020-05-19:
I am trying to test the 40Gb HSSI MAC/PHY IP example with `hssi_loopback` utility following the document here:  https://opae.github.io/latest/docs/fpga_tools/hssi_loopback/readme.html. `hssi_loopback` was able to find the e40 HSSI interface successfully but failed to initialize the AFU:
```shell
$ fpgaconf eth_e2e_e40.gbs
$ hssi_loopback readmacs -B 0xAF
Found e40
Error initializing accelerator
```
The bitstream is compiled with Quarts 17.1 and Intel Acceleration Stack version 1.2.1. And I am using Arria-10 on Intel vLab `fpga-pac-a10` node. 

Also it seems that the latest OPAE-SDK does not provide any other testing utilities for HSSI device other than the [pac_hssi_config.py](https://github.com/OPAE/opae-sdk/tree/master/tools/extra/pac_hssi_config)? The `hssi_config` is moved to legacy repo here: https://github.com/OPAE/opae-legacy/blob/master/tools/hssi/. The `pac_hssi_config.py` does not work in my case since I do not the the root access to open the `/dev/mem`

```shell
$ pac_hssi_config.py e40init 18:00.0
Traceback (most recent call last):
  File "./pac_hssi_config.py", line 1184, in <module>
    main()
  File "./pac_hssi_config.py", line 1168, in main
    mem.open(get_bdf_base_addr(args.bdf, 0), 0x80000)
  File "./pac_hssi_config.py", line 62, in open
    self.fdesc = os.open("/dev/mem", os.O_RDWR)
PermissionError: [Errno 13] Permission denied: '/dev/mem'
```

tswhison, 2020-05-19:
Hi Hecmay, thanks for you query.  The hssi tools that were moved to opae-legacy are not compatible with the e40 device.  Those tools correspond to an old design that is no longer offered.  The hssi tools that are currently found in the opae-sdk repo are for the current hardware design.

Hecmay, 2020-05-19:
@tswhison Thanks for your reply.

But is there a way to configure it without requiring sudo access? I always get a permission error from the new hssi config tools reading the `/dev/mem`.