OFS/opae-sdk #1583: Support status of D5005 PAC

ddcc, 2020-06-15:
I've recently been testing the D5005 PAC, and have noticed there seems to be some differences in support status compared to the Arria 10. All of the release notes on GitHub only mention the N3000 or the Arria 10, and not the D5005. Additionally, the upstream driver doesn't even have the PCI ID 0b2b for the D5005 at all. Does this mean that this card is only supported by the legacy driver, of which the latest version is 2.0.4?

tswhison, 2020-06-15:
Hi and thanks for your question.  We are in the process of converging on a single software base that will support A10, N3000, and D5005, via the up-streaming effort.  I can't speak to specific timeframes, but the work is in progress.

ddcc, 2020-06-15:
Thanks for the clarifying. The reason I'm asking is that I'm running into a weird issue, where after a few minutes of uptime, the D5005 (v2.0.2) just completely stops responding and seems to drop off the PCIe bus. The green DONE LED on the back goes dark, the interface UUID is reported as all 0xff, and even `lspci` doesn't show anything:

```
01:00.0 Processing accelerators: Intel Corporation Device 0b2b (rev ff) (prog-if ff)
!!! Unknown header type 7f
Kernel driver in use: intel-fpga-pci
Kernel modules: intel_fpga_pci
```

Even if I reboot the machine, the PAC remains unresponsive, and to the remainder of the system, it appears as if nothing is physically present in the PCIe slot, both in Linux and BIOS. It isn't until I physically unplug the machine for a while that the PAC card shows up again.

southerngs, 2020-06-15:
When I've seen this problem reported by other people the issue has been that FPGA overheated.  You might be able to check temperature with `fpgainfo temp`.

ddcc, 2020-06-15:
Ok, I will keep that in mind. I did switch the D5005 back out for the Arria 10, so I can't check right now, but I do recall it being a bit warm. The overheating would be unfortunate, because it's been occurring just a few minutes after the machine boots, so I haven't even uploaded an AFU yet.

zhipengzhaocmu, 2020-06-17:
Hi I got the same issue. The D5005 worked normally for a few minutes and then stopped working. When the D5005 worked normally, I got expected results by `lspci `and `fpgainfo fme`. After 4 minutes, the `lspci ` returns `!!! Unknown header type 7f`, and the `fpgainfo fme` got all `0xFF` in Bitstream Id and Pr Interface Id. 

I tried `fpgainfo temp`,  but got some errors.
```
metrics_max10.c:191:dfl_enum_max10_metrics_info() **ERROR** : Failed pattern match /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/dfl-fme*/spi-altera*/spi_master/spi*/spi*/*bmc-hwmon*/hwmon/hwmon*: No such file or directory
metrics_utils.c:590:enum_fpga_metrics() **ERROR** : Failed to Enum Power and Thermal metrics.
metrics.c:78:xfpga_fpgaGetNumMetrics() **ERROR** : Failed to Discover Metrics
Error getting number of metrics: not found
Error reading metrics from BMC: not found 
```

ddcc, 2020-06-17:
To provide some more context, we've been testing different D5005 cards on different machines. Mine is a MSI MS-7B50 with an Intel i9-9900k, and I've tried both Ubuntu 19.10 with kernel 5.3.0-59 and Ubuntu 20.04 with kernel 5.4.0-37. Zhipeng has an ASRock X299 Taichi with an Intel i7-7820X, running Ubuntu 16.04 with Linux kernel 4.15.18.

I think the only thing in common is that we've both been using opae-intel-fpga-driver v2.0.4-2, the latest OPAE from GitHub, and Acceleration Stack v2.0.1. Also, we've noticed that if we remove other PCIe devices from the system (e.g. GPU, NIC, etc), the D5005 seems to work normally for longer, but still only the order of minutes.

southerngs, 2020-06-17:
@zhipengzhaocmu did you run `fpgainfo temp` before or after the card stopped working?

The D5005 is designed for use in rack mounted servers where the server fans provides cooling for passively cooled PCI cards.  

In our experience when the D5005 is used in a desktop system we also needed to add auxiliary cooling.

zhipengzhaocmu, 2020-06-17:
I run `sudo fpgainfo temp` before the card stopped working.
```
board_d5005.c:72:read_bmcfw_version() **ERROR** : Failed to get read object
board_d5005.c:211:print_board_info() **ERROR** : Failed to read bmc version
board_d5005.c:132:read_max10fw_version() **ERROR** : Failed to get read object
board_d5005.c:217:print_board_info() **ERROR** : Failed to read max10 version
Board Management Controller, MAX10 NIOS FW version:
Board Management Controller, MAX10 Build version:
//****** TEMP ******//
Object Id                        : 0xF100000
PCIe s:b:d.f                     : 0000:65:00.0
Device Id                        : 0x0B2B
Socket Id                        : 0x00
Ports Num                        : 01
Bitstream Id                     : 0x202000200000237
Bitstream Version                : 2.0.2
Pr Interface Id                  : 9346116d-a52d-5ca8-b06a-a9a389ef7c8d
Boot Page                        : user
metrics_max10.c:191:dfl_enum_max10_metrics_info() **ERROR** : Failed pattern match /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/dfl-fme*/spi-altera*/spi_master/spi*/spi*/*bmc-hwmon*/hwmon/hwmon*: No such file or directory
metrics_utils.c:590:enum_fpga_metrics() **ERROR** : Failed to Enum Power and Thermal metrics.
metrics.c:78:xfpga_fpgaGetNumMetrics() **ERROR** : Failed to Discover Metrics
Error getting number of metrics: not found
Error reading metrics from BMC: not found
```

I did use an auxiliary 20-in box fan for better cooling. The thermometer attached to the machine case reports 83F. 
Would it be a power issue? The power supply for the desktop is 650W. Is that enough?

southerngs, 2020-06-17:
I doubt that it is power supply issue, 650W should be plenty.  I still suspect thermal issue, that would be consistent with FPGA working initially and then shutting off after the board warms up.

The fpgainfo error looks like possible bug, but it could be the result of using "unsupported" configuration or mismatch between driver development and what is on OPAE branch.  I'm not working on OPAE currently so I don't know what driver and SDK combination is needed for D5005.

You may want to test using the software that shipped with D5005.  I know you are on Ubuntu so you don't have deb packages but I think you can install from source for OPAE SDK (the driver might require converting deb to rpm).  That would at least let you verify if your card is working correctly (at boot) and see what temperature is reported before it stops responding.

tswhison, 2020-06-17:
The fpgainfo errors are consistent with a driver / user space code mismatch.  Please try both the driver and user mode that were provided for the specific card.

ddcc, 2020-06-17:
Thanks for the follow-ups, I plan to swap the cards back out and retry with additional cooling later today.

It seems like the fpgainfo errors are a separate issue. Yeah, the issue is that the v2.0.1 installer for D5005 only includes Red Hat packages, which aren't compatible with Ubuntu. It looks like the provided versions are 2.0.2 for the driver, and 1.1.4 for OPAE.

ddcc, 2020-06-18:
I can confirm that at least in my case, the issue was overheating all long. Initially, I tried just pointing an external fan at the side of my machine, but it didn't make a difference until I removed the black metal shroud from the FPGA.

After starting my experiments, I made some temperature measurements of the FPGA under load, with an IR camera. They show that even with significant external airflow, the side of the heatsink is reaching 65 C, so the underlying FPGA is likely getting even hotter.

zhipengzhaocmu, 2020-06-25:
Forgot to mention. My D5005 is also an overheating problem. After removing the black metal shroud and having a fan point to the board, it works fine. Thanks for your help!

quantgeek, 2020-11-12:
Can someone help me install D5005 on Ubuntu 20.1?  I am not sure what needs to be installed to get it working.  The stack seems to be targeted to Redhat and thus uses an older version of the kernel.  I suspect I just install the SDK as the driver is already in the kernel?  Thanks for your help!

zhipengzhaocmu, 2020-11-12:
> Can someone help me install D5005 on Ubuntu 20.1? I am not sure what needs to be installed to get it working. The stack seems to be targeted to Redhat and thus uses an older version of the kernel. I suspect I just install the SDK as the driver is already in the kernel? Thanks for your help!

I have installed it on Ubuntu 16.04. Here are my steps:
1. Follow the UG to install the Intel Acceleration Stack Development Package on the Host Machine (I used 2.0.1 version)
```
tar xvf *dev_installer.tar.gz
cd *dev_installer
```
change "is_ubuntu" to 1 in setup.sh
`./setup.sh`
Accept the license.
Skip the OPAE
`source /home/<username>/inteldevstack/init_env.sh`

2. Install OPAE
```
git clone https://github.com/OPAE/opae-sdk.git
cd opae-sdk
mkdir build
cd build
cmake -DCPACK_GENERATOR=DEB ..
make
make package_deb  
sudo dpkg -i  opae*.deb
```
3. Install driver (download the opae-intel-fpga-driver-2.0.4-2.x86_64.rpm from  https://github.com/OPAE/opae-sdk/releases/download/1.4.0-1/opae-intel-fpga-driver-2.0.4-2.x86_64.rpm)
```
sudo alien opae-intel-fpga-driver-2.0.4-2.x86_64.rpm
sudo dpkg -i /home/zzhao1/pac_d5005/opae-intel-fpga-driver-2.0.4-3_amd64.deb
sudo dkms install opae-intel-fpga-driver/2.0.4-2
```
```
dkms status
```

4. dump fpga info
`sudo fpgainfo fme`
5. Run Hello_fpga
```
sudo fpgaconf $OPAE_PLATFORM_ROOT/hw/samples/nlb_mode_0/bin/nlb_mode_0_unsigned.gbs
sudo sh -c "echo 20 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages"
```
Compile the hello_fpga(https://opae.github.io/0.9.0/docs/fpga_api/quick_start/readme.html)
```
cd  opae-sdk/samples
gcc -std=c99 hello_fpga.c -I/usr/local/include -L/usr/local/lib -lopae-c -luuid -ljson-c -lpthread -o hello_fpga
sudo ./hello_fpga
```

quantgeek, 2020-12-06:
@zhipengzhaocmu: Thank you for this response.  If I install this, will I also be able to use the Arria 10 GX as well?  I suspect this will succeed on Ubuntu 20?

@ddcc: Any chance you can post your installation guide?

Thanks,

QG

ddcc, 2020-12-06:
Zhipeng has a similar setup as me, except that I'm on Ubuntu 20.04, and used alien to convert from rpm to deb. You can also try using the upstream in-kernel driver, though last time I tried, it didn't support shared FPGA access. You need different Acceleration SDK versions for the Arria 10 and D5005, so you'd need to have both in separate locations on the filesystem, and switch between them when building your bitstream.

atbolsh, 2022-06-16:
> I have installed it on Ubuntu 16.04. Here are my steps:
>
>     1. Follow the UG to install the Intel Acceleration Stack Development Package on the Host Machine (I used 2.0.1 version)
>
>
> ```
> tar xvf [...]

Hi, this is a great guide. However, in my case I fail half way through, I think because I have an older FPGA, and hence an older version of the OPAE.

Specifically, I have an Arria 10. This is what the UG says I should install for the OPAE:
https://www.intel.com/content/www/us/en/software-kit/665889/intel-pac-with-intel-arria-10-gx-fpga-aceleration-stack-version-1-2.html?swcode=WWW-SWD-IAS-A10-GX-PAC-PV-DEV-121

When I try to install it using setup.sh (changing is_ubuntu to 1), I get errors because it tries to install packages that don't exist (libjson0 ). When I fix those lines to point to the modern versions of those packages, it ends up failing later:

Building module:
cleaning build area...
make -j32 KERNELRELEASE=5.13.0-48-generic KERNELDIR=/lib/modules/5.13.0-48-generic/build....(bad exit status: 2)
ERROR (dkms apport): unable to determine source package for opae-intel-fpga-driver
Error! Bad return status for module build on kernel: 5.13.0-48-generic (x86_64)
Consult /var/lib/dkms/intel-fpga/1.1.2-1/build/make.log for more information.

-------------------------------------------------------------------------------
- installing OPAE software ...
-------------------------------------------------------------------------------
>>> Running cmd:
      sudo dpkg -i opae-libs-1.1.2-1.x86_64.deb

Selecting previously unselected package opae-libs.
(Reading database ... 211279 files and directories currently installed.)
Preparing to unpack opae-libs-1.1.2-1.x86_64.deb ...
Unpacking opae-libs (1.1.2) ...
dpkg: dependency problems prevent configuration of opae-libs:
 opae-libs depends on libjson0; however:
  Package libjson0 is not installed.

dpkg: error processing package opae-libs (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 opae-libs
Command: "sudo dpkg -i opae-libs-1.1.2-1.x86_64.deb" exited with error code: 1

Any advice?

SiamakGit, 2024-04-25:
> I've recently been testing the D5005 PAC, and have noticed there seems to be some differences in support status compared to the Arria 10. All of the release notes on GitHub only mention the N3000 or [...]

> I've recently been testing the D5005 PAC, and have noticed there seems to be some differences in support status compared to the Arria 10. All of the release notes on GitHub only mention the N3000 or [...]

Hi Dominic,I am very new in PAC hardwares.could you please tell me how can I extract the pin assignments? I just want to blink one LED from the board.I searched everywhere for pin assignment for the FPGA 
and I cant find it?