OFS/opae-sdk #1585: Arria 10 PAC won't update due to SPI flash timeout

ddcc, 2020-06-15:
I'm trying to apply the 1.2.1 update, but it is timing out for me:

```
(python) # fpgaotsu /usr/share/opae/a10-gx-pac/fpgaotsu/base/otsu-09C4.json
[2020-06-15 12:59:13,298] [INFO    ] [MainThread] Intel PAC with Intel Arria 10 GX FPGA 0000:01:00.0 is not secure.
[2020-06-15 12:59:13,307] [WARNING ] [MainThread] Update starting. Please do not interrupt.
[2020-06-15 12:59:13,307] [INFO    ] [0000:01:00.0] Updating Intel PAC with Intel Arria 10 GX FPGA : 0000:01:00.0
[2020-06-15 12:59:13,307] [INFO    ] [0000:01:00.0] Erasing flash@0x7fd0000 for 196608 bytes
[2020-06-15 12:59:33,313] [ERROR   ] [0000:01:00.0] [Errno 110] Connection timed out
Traceback (most recent call last):
  File "/home/ddcc/opae-sdk/python/opae.admin/opae/admin/tools/fpgaotsu.py", line 511, in process_flash_item
    self.erase(flash, mtd_dev)
  File "/home/ddcc/opae-sdk/python/opae.admin/opae/admin/tools/fpgaotsu.py", line 317, in erase
    mtd_dev.erase(erase_start,
  File "/home/ddcc/opae-sdk/python/opae.admin/opae/admin/utils/mtd.py", line150, in erase
    fcntl.ioctl(fp.fileno(), self.IOCTL_MTD_MEMERASE, iodata)
TimeoutError: [Errno 110] Connection timed out
[2020-06-15 12:59:33,314] [INFO    ] [MainThread] Total time: 0:00:10.006953
[2020-06-15 12:59:33,314] [ERROR   ] [MainThread] Intel PAC with Intel Arria 10 GX FPGA 0000:01:00.0: [Errno 110] Connection timed out
[2020-06-15 12:59:33,314] [ERROR   ] [MainThread] One-Time Secure Update failed
```

When I examine the kernel log, it appears the SPI flash driver is timing out:

```
[ 3696.799549] altr-asmip2 altr-asmip2.1.auto: altera_asmip2_read_reg timeout
[ 3696.799551] error -110 reading SR
```

ddcc, 2020-06-15:
I did come across some comments from Matthew Gerlach that this SPI flash driver has a 10 sec timeout to prevent the system from hanging: https://patchwork.kernel.org/patch/9962009/#21027297

fpgamatt, 2020-06-16:
From the output of fpgaotsu and the kernel message, it looks like fpgaotsu is trying to erase a sector of flash.  Erasing flash involves sending a "flash erase" command to the flash chip and then polling on the SR register in the flash looking for the Work In Progress bit to clear.  It looks like the flash chip is not responding to the SR register request.  

This could be a temperature problem or flash problem.

ddcc, 2020-06-16:
Hmm, ok. Do the following temperatures look normal? 

```
Board Management Controller, microcontroller FW version: 26889
Last Power down cause:POK_CORE
Last Reset cause: None
//****** TEMP ******//
Object Id                        : 0xF000000
PCIe s:b:d.f                     : 0000:01:00.0
Device Id                        : 0x09C4
Socket Id                        : 0x00
Ports Num                        : 01
Bitstream Id                     : 0x123000200000185
Bitstream Version                : 1.2.3
Pr Interface Id                  : 69528db6-eb31-577a-8c36-68f9faa081f6
Boot Page                        : user
( 1) FPGA Core TEMP              : 62.00 Centigrade
( 2) Board TEMP                  : 52.00 Centigrade
( 3) QSFP TEMP                   : N/A
( 4) Core Supply Temp            : 68.84 Centigrade
```

jyoung3131, 2020-09-03:
Hi all - we have this same problem on at least one of our two lcaol Arria10 PAC cards. 

Card 1 - fails during the initial fgaotsu update: 
```
[]$ sudo fpgaotsu /usr/share/opae/a10-gx-pac/fpgaotsu/base/otsu-09C4.json
[2020-09-02 21:12:14,385] [INFO    ] [MainThread] Intel PAC with Intel Arria 10 GX FPGA 0000:01:00.0 is not secure.
[2020-09-02 21:12:14,393] [WARNING ] [MainThread] Update starting. Please do not interrupt.
[2020-09-02 21:12:14,394] [INFO    ] [0000:01:00.0] Updating Intel PAC with Intel Arria 10 GX FPGA : 0000:01:00.0
[2020-09-02 21:12:14,396] [INFO    ] [0000:01:00.0] Erasing flash@0x7fd0000 for 196608 bytes
[2020-09-02 21:12:14,591] [INFO    ] [0000:01:00.0] Writing flash@0x7ff0000 for 104 bytes (RushCreek_Release_key_blk2)
(100%) [████████████████████] [104/104 bytes][Time:0:00:00.000068]              
[2020-09-02 21:12:14,593] [INFO    ] [0000:01:00.0] Reading flash@0x7ff0000 for 104 bytes for verification
(100%) [████████████████████] [104/104 bytes][Time:0:00:00.006021]              
[2020-09-02 21:12:14,600] [INFO    ] [0000:01:00.0] Verified flash@0x7ff0000 for 104 bytes (RushCreek_Release_key_blk2)
[2020-09-02 21:12:14,600] [INFO    ] [0000:01:00.0] Erasing flash@0x1800000 for 41943040 bytes
[2020-09-02 21:12:54,742] [INFO    ] [0000:01:00.0] Writing flash@0x1800000 for 41943040 bytes (dcp_1_2_1_rot_reversed.rpd)
(100%) [████████████████████] [41943040/41943040 bytes][Time:0:01:06.238681]    
[2020-09-02 21:14:01,045] [INFO    ] [0000:01:00.0] Reading flash@0x1800000 for 41943040 bytes for verification
(100%) [████████████████████] [41943040/41943040 bytes][Time:0:00:58.330970]    
[2020-09-02 21:14:59,404] [INFO    ] [MainThread] Total time: 0:02:45.010901
[2020-09-02 21:14:59,405] [ERROR   ] [MainThread] Intel PAC with Intel Arria 10 GX FPGA 0000:01:00.0: Verification of dcp_1_2_1_rot_reversed.rpd @0x25165824 failed
[2020-09-02 21:14:59,405] [ERROR   ] [MainThread] One-Time Secure Update failed
```

Card 1 temps:
```
fpgainfo temp
Board Management Controller, microcontroller FW version 26889
Last Power Down Cause: POK_CORE
Last Reset Cause: None
//****** TEMP ******//
Object Id                     : 0xEE00000
PCIe s:b:d:f                  : 0000:01:00:0
Device Id                     : 0x09C4
Socket Id                     : 0x00
Ports Num                     : 01
Bitstream Id                  : 0x123000200000185
Bitstream Version             : 1.2.3
Pr Interface Id               : 69528db6-eb31-577a-8c36-68f9faa081f6
Boot Page                     : user
(11) FPGA Core TEMP           : 38.00 °C
(12) Board TEMP               : 32.00 °C
(14) QSFP TEMP                : No reading (reading state unavailable)
(15) Core Supply Temp         : 43.92 °C
```

closed 2021-10-11 by tswhison