OFS/opae-sdk #1697: Unable to build from source - CLI11 integration?

gnbond, 2020-09-25:
I'm attempting to build the SDK (in particular the tools) from source, as shown [here](https://opae.github.io/2.0.0/docs/fpga_api/quick_start/readme.html#building-and-installing-the-opae-sdk-from-source)

This is failing in the cmake -> CLI11 portion

```
Error copying file "/home/gregory-bond/Workspace/OPAE/opae-sdk/external/CLI11/download/CLI11-prefix/src/CLI11-stamp/CLI11-gitinfo.txt" to "/home/gregory-bond/Workspace/OPAE/opae-sdk/external/CLI11/download/CLI11-prefix/src/CLI11-stamp/CLI11-gitclone-lastrun.txt".
CMake Error at CLI11/download/CLI11-prefix/tmp/CLI11-gitclone.cmake:106 (message):
  Failed to copy script-last-run stamp file:
  '/home/gregory-bond/Workspace/OPAE/opae-sdk/external/CLI11/download/CLI11-prefix/src/CLI11-stamp/CLI11-gitclone-lastrun.txt'

```

I see the same behaviour with HEAD, 2.0.0-1, and 1.5.0-2

I'm running Ubuntu 18 but I don't think this is distro-specific.

r-rojo, 2020-09-28:
Does your environment have access to the Internet?
Are you behind a firewall or proxy that may be restricting your access?
Can you also check if you've run out of disk space or disk quota (if enabled).
I just tried right now with Ubuntu 18 and with Ubuntu 20 and didn't have any problems.
Can you try again and see if you still get the error?
I've run into something similar downloading externals before but it worked a few minutes later.
I can only guess it may have been a small outage with github at that time.

gnbond, 2020-09-29:
Thanks for your comments.  No problems with disk space or internet links.

However I removed the entire project and started again from "git clone" and this time it's worked OK.  Happy to close this and attribute to the many wonders of the Internet.

closed 2020-10-02 by r-rojo