OFS/opae-sdk #1710: Ffailure building intel OPAE PAC driver for kernel 5.8.0-dfl

tinebp, 2020-10-20:
System: Ubuntu 18.04

DKMS make.log for intel-fpga-2.0.3 for kernel 5.8.0-dfl (x86_64)
Tue Oct 20 11:15:03 EDT 2020
make -C /lib/modules/5.8.0-dfl/build M=/var/lib/dkms/intel-fpga/2.0.3/build modules
make[1]: Entering directory '/localscratch/btine3/linux-dfl-2.0.0-1'
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/i2c/busses/i2c-altera.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/intel-generic-qspi.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/devices/intel-on-chip-flash.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/altera-asmip2.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/spi/spi-altera-4-12.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/spi-nor-4-12.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mfd/intel-max10-main.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mfd/intel-max10-debug.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mfd/intel-spi-avmm.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mfd/fdt_reg.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt_ro.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt_wip.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt_rw.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt_sw.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt_strerror.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/lib-4-12/fdt_empty_tree.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/misc/avmmi-bmc.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/fpga-mgr.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/pcie.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/pcie_check.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/feature-dev.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-pr.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-tcm.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-tcm-debug.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-iperf.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-dperf.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-error.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-eth.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/afu.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/region.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/dma-region.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/afu-error.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/afu-check.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/pac-hssi.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/pac-hssi-multi-qsfp.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/intel-fpga-pac-iopll.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/net/ethernet/pac_n3000_net.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/net/ethernet/c827_retimer.o
  CC [M]  /var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/ifpga-sec-mgr.o
In file included from ./include/linux/regmap.h:20:0,
                 from /var/lib/dkms/intel-fpga/2.0.3/build/include/linux/regmap-intel-pac.h:5,
                 from /var/lib/dkms/intel-fpga/2.0.3/build/include/linux/mtd/intel-generic-qspi.h:12,
                 from /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/intel-generic-qspi.c:15:
./include/linux/iopoll.h:120:0: warning: "readx_poll_timeout" redefined
 #define readx_poll_timeout(op, addr, val, cond, sleep_us, timeout_us) \
 
In file included from /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/intel-generic-qspi.c:13:0:
/var/lib/dkms/intel-fpga/2.0.3/build/include/linux/iopoll-4-12.h:43:0: note: this is the location of the previous definition
 #define readx_poll_timeout(op, addr, val, cond, sleep_us, timeout_us) \
 
In file included from ./include/linux/regmap.h:20:0,
                 from /var/lib/dkms/intel-fpga/2.0.3/build/include/linux/regmap-intel-pac.h:5,
                 from /var/lib/dkms/intel-fpga/2.0.3/build/include/linux/mtd/intel-generic-qspi.h:12,
                 from /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/intel-generic-qspi.c:15:
./include/linux/iopoll.h:140:0: warning: "readx_poll_timeout_atomic" redefined
 #define readx_poll_timeout_atomic(op, addr, val, cond, delay_us, timeout_us) \
 
In file included from /var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/intel-generic-qspi.c:13:0:
/var/lib/dkms/intel-fpga/2.0.3/build/include/linux/iopoll-4-12.h:78:0: note: this is the location of the previous definition
 #define readx_poll_timeout_atomic(op, addr, val, cond, delay_us, timeout_us) \
 
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/mtd/spi-nor/spi-nor-4-12.c: In function ‘set_4byte’:
[... 470 lines cut ...]
  pdata.regbus = indirect_regbus;
       ^
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.c:1152:39: error: ‘ALTERA_SPI_DRV_NAME’ undeclared (first use in this function); did you mean ‘ALTERA_I2C_DRV_NAME’?
  subdev = feature_create_subdev(pdev, ALTERA_SPI_DRV_NAME, &pdata,
                                       ^~~~~~~~~~~~~~~~~~~
                                       ALTERA_I2C_DRV_NAME
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.c:1152:39: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.c: In function ‘nios_spi_init’:
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.c:1635:7: error: ‘struct altera_spi_platform_data’ has no member named ‘regmap_ctx’
  pdata.regmap_ctx = base;
       ^
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.c:1636:7: error: ‘struct altera_spi_platform_data’ has no member named ‘regbus’
  pdata.regbus = nios_indirect_regbus;
       ^
/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.c:1642:39: error: ‘ALTERA_SPI_DRV_NAME’ undeclared (first use in this function); did you mean ‘ALTERA_I2C_DRV_NAME’?
  subdev = feature_create_subdev(pdev, ALTERA_SPI_DRV_NAME, &pdata,
                                       ^~~~~~~~~~~~~~~~~~~
                                       ALTERA_I2C_DRV_NAME
scripts/Makefile.build:280: recipe for target '/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.o' failed
make[2]: *** [/var/lib/dkms/intel-fpga/2.0.3/build/drivers/fpga/intel/fme-main.o] Error 1
Makefile:1756: recipe for target '/var/lib/dkms/intel-fpga/2.0.3/build' failed
make[1]: *** [/var/lib/dkms/intel-fpga/2.0.3/build] Error 2
make[1]: Leaving directory '/localscratch/btine3/linux-dfl-2.0.0-1'
Makefile:115: recipe for target 'all' failed
make: *** [all] Error 2

rweight, 2020-10-21:
If you are running a 5.8.0-dfl kernel, I'm assuming you are using the kernel from: https://github.com/OPAE/linux-dfl/tree/fpga-ofs-dev-5.8.0?

There are two different versions of the driver code: the intel-fpga driver, and the linux-dfl driver.

(1) The linux-dfl driver is the newer version that is being actively upstreamed. Much of the driver code has already been accepted. The fpga-ofs-dev-5.8.0 branch consists of the 5.8.0 linux kernel with the dfl driver integrated into the kernel.

(2) The intel-fpga driver is the legacy version of the driver. It is an out-of-tree driver that is supported for specific kernel versions (Centos 7, LTS 4.19)

If you want to use the intel-fpga driver, I would recommend installing it on a CentOS 7 system, optionally using the 4.19 kernel.

If you want too use the linux-dfl driver, then if you are running the 5.8.0-dfl kernel, it is already installed and running. Please refer to the wiki pages associated with the linux-dfl repository for more information on this version of the driver.

stefanstoeckigt, 2021-05-05:
@rweight do you know when the driver will be in the upstream kernel?