OFS/opae-sdk #1754: D5005 PAC doesn't show up on PCIe

sidsahay, 2020-11-08:
I can't seem to get one of our D5005s to show up on PCIe or `fpgainfo` regardless of the configuration/system I try it in. The 12V power LED is on, but the Power Good LED and the AFU LEDs are always off i.e. only one orange LED is on when the machine boots up. 

We know the machine supports the card because it works fine with a different D5005, which has 12V power, power good, and done LEDs light up when it boots and it works fine with the FPGA tools and OPAE, so we suspect there's something wrong with this particular card. Any ideas what could be the issue? I would have tried reflashing the BMC but can't since the card doesn't show up in the system.

GrueMaster, 2021-01-11:
If the card doesn't show up in lspci, this is not a software issue.  You should contact your sales rep or customer support to see about having that card serviced.

sidsahay, 2021-01-14:
Thanks, will do.

closed 2021-01-14 by sidsahay