OFS/opae-sdk #1898: Can't install PACSign app.

saitej25, 2021-03-03:
How can I install this utility?

saitej25, 2021-03-03:
Tried 
cd opae-sdk/python/pacsign/
python3 ./setup.py build
python3 ./setup.py install

chrbirks, 2021-03-05:
That method has worked for me before except for running the last install command with sudo. But it's hard to say without any error messages.

r-rojo, 2021-03-05:
Like @chrbirks said, it's hard to know what could be wrong without more information, like error messages.
Can you provide which Ubuntu distro you're trying this on as well as the Python version?
It could be that your user does not have sufficient privileges to install files into that Python instance.
You could also use "python3 -m pip install <path to PACSign>  --user" to install it into your user's Python site-packages directory.

saitej25, 2021-03-09:
works with sudo

closed 2021-03-08 by saitej25