OFS/opae-sdk #1994: fpgasupdate.py wont work on RO file system

rmr-silicom, 2021-04-12:
When the filesystem is RO, the following fails.

os.mkdir(intel_fw_path, 0o755)

Why does it need to be in /usr/lib/firmware?

rmr-silicom, 2021-04-13:
I figured this out. But the script still won't work on a read only file system.

echo -n /root > /sys/module/firmware_class/parameters/path
echo -n ofs_fim_page1_unsigned.bin > /sys/class/fpga_sec_mgr/fpga_sec0/update/filename

I tried bind mounting, but it didn't really work (RHCOS)...

tswhison, 2021-04-14:
Yes, that's right.  The directory /usr/lib/firmware/intel must exist (or else the script will attempt to create it) and it must be writable.  We are wondering whether your desired workflow is for test or for production?  If for test, there are a couple of workarounds you might consider.

Workaround 1) Using sysfs directly.  With this approach, you could create symbolic links in /usr/lib/firmware/intel, such as /usr/lib/firmware/intel/myfw.bin.  The myfw.bin link could point to a file that is on an RW filesystem.

Workaround 2) Enabling fpgasupdate.  With this approach, you could rename any existing /usr/lib/firmware/intel directory to, say intel-save and then re-create the intel directory under /usr/lib/firmware as a symbolic link to a directory on a RW filesystem.  In this case, you would be able to run fpgasupdate to update the device, because the script would be able to create the temporary files.

Note that this restriction is due to the kernel's firmware APIs that are being used by the secure manager driver, and that these drivers are being upstreamed; so, they are unlikely to change.

rmr-silicom, 2021-04-15:
Thanks for the reply!

This is for production to program an FPGA PCIe card.

However, it is not possible to create any files under /usr in RHCOS (in an Openshift cluster) using the ignition tool. I have used a bind mount to circumvent this type of issue before (RO FS), but for some reason it didn't seem to work.

I will stick with the sysfs solution for now.

closed 2021-04-19 by rmr-silicom