OFS/opae-sdk #2410: FPGA device

shanshuixiangyi, 2021-12-02:
Hello ,,,
   I am using Intel ® Stratix ® 10 MX FPGA Development Kit for ETH miner. Can i computer use this OPAE software to communicate with my FPGA board to accelerate my ethash algorithm ?

Thanks.

tswhison, 2021-12-06:
Hello, thank you for your question.  The short answer is, it depends on whether your PCI Device ID is supported.  If you are creating your FIM IP based one a reference design, then it should be supported.  However, if you deviate from the reference design or implement a PCIe DID that the OPAE SDK source does not track, then you may run into issues.  In this case, you would need to modify the OPAE SDK to make it understand your DID.

nathanborak, 2024-10-14:
Closing due to inactivity/age, and because the original question was answered. If you still require assistance, please re-open the issue.

closed 2024-10-14 by nathanborak