OFS/opae-sdk #2411: fpgaotsu and fpgasupdate are not found on ubuntu 20.04

sungin-h, 2021-12-02:
Hi, I'm new to using Intel FPGA PAC.
I'm trying to install PAC D5005 on ubuntu 20.4 and Acceleration Stack. But, I can't find 'fpgaostu' and 'fpgasupdate' program.
(I know ubuntu is not officially supported, but I need it.)

I downloaded Acceleration Stack Version 2.0.1 and installed using 'setup.sh' script.
Before running the script, I changed 'is_ubuntu' to 1(encouraged by [this link](https://github.com/OPAE/opae-sdk/issues/1583#issuecomment-726155989) ). Also, I inserted a few lines for converting *.rpm to *.deb using 'alien' program. 

To verify the installation, I checked the driver installation, OPAE library installation, FIM/BMC version using fpgainfo.
$sudo lsmod  | grep fpga
$sudo dpkg -l | grep opae
$sudo fpgainfo fme

The above tests can be passed.

However, some programs are missing.
I can find:
   fpgabist fpgaconf fpga_crossgen fpgad fpgadiag fpga_dma_test fpgainfo fpga_libtool fpgamux
I can't find:
   fpgaotsu fpgasupdate

Where can I find those programs or What package do I miss/fail to install?

thank you.

tswhison, 2021-12-02:
You could use
```bash
$ pip install opae-sdk/python/opae.admin
```
which should give you those scripts.  They are part of the opae.admin python project.

sungin-h, 2021-12-03:
@tswhison thank you. successfully installed.

tswhison, 2021-12-06:
Filer was able to install the requested scripts.

closed 2021-12-06 by tswhison

yli147, 2024-10-28:
did you get the intel fpga driver installed on ubuntu 20.04 successfully ?