OFS/opae-sdk #2514: Issue with OPAE-SDK

knmurth, 2022-03-11:
Hi all,
I have build and compiled the newest available version of OPAE-SDK on a CPU which runs ubuntu 21.10.I am encountering an error while running fpgainfo fme 
[image]
Kindly propose any solution for the above problem

tswhison, 2022-03-14:
Hi @knmurth, do you have the DFL drivers loaded on the system prior to running fpgainfo?

knmurth, 2022-03-16:
Hi@tswhison,I have loaded OPAE-SDK,I am inheriting the tar-files of OPAE-Intel FPGA Drivers which are available in the assets of OPAE-SDK git repository.Can you provide a path to how to load DFL drivers for ubuntu 21.10LTS system

nathanborak, 2024-10-14:
Closing due to inactivity/age, and because the original question was answered. If you still require assistance, please re-open the issue.

closed 2024-10-14 by nathanborak