OFS/opae-sdk #2557: rpmbuild fails on Fedora 32

bdelgado1995, 2022-04-16:
I'm following the instructions at https://opae.github.io/latest/docs/install_guide/installation_guide.html and trying to build the rpms for Fedora 32. 

When running this command, I get the errors below. 

$ cd opae-sdk/packaging/opae/rpm
$ ./create fedora

Errors: 

Checking for unpackaged file(s): /usr/lib/rpm/check-files /home/brian/opae-sdk/packaging/opae/rpm/rpmbuild/BUILDROOT/opae-2.1.0-1.fc32.x86_64
error: Installed (but unpackaged) file(s) found:
   /usr/lib/opae-2.1.0/opae-config-version.cmake
   /usr/lib/opae-2.1.0/opae-config.cmake
   /usr/lib/opae-2.1.0/opae-targets-relwithdebinfo.cmake
   /usr/lib/opae-2.1.0/opae-targets.cmake

RPM build errors:
    File listed twice: /usr/lib/.build-id/b8/e0e176bef6cc46ac0d9d542613d3eef9108e90
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/PKG-INFO
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/SOURCES.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/dependency_links.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/entry_points.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/requires.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/top_level.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__init__.py
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__pycache__
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__pycache__/__init__.cpython-38.opt-1.pyc
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__pycache__/__init__.cpython-38.pyc
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/_opae.cpython-38-x86_64-linux-gnu.so
    Installed (but unpackaged) file(s) found:
   /usr/lib/opae-2.1.0/opae-config-version.cmake
   /usr/lib/opae-2.1.0/opae-config.cmake
   /usr/lib/opae-2.1.0/opae-targets-relwithdebinfo.cmake
   /usr/lib/opae-2.1.0/opae-targets.cmake
first rpmbuild failed.. exiting

r-rojo, 2022-04-19:
#2559 Should fix this

bdelgado1995, 2022-04-19:
Thanks for the update! I tried it out but am now getting a slightly similar error, but not exactly the same. If it's helpful, I can attach the full log. 

git rev-parse HEAD
8e879d158a60e09427338dc72fd8bbdd51deb264

./create fedora 
[... ]
Checking for unpackaged file(s): /usr/lib/rpm/check-files /home/brian/opae-sdk/packaging/opae/rpm/rpmbuild/BUILDROOT/opae-2.1.0-1.fc32.x86_64
error: Installed (but unpackaged) file(s) found:
   /usr/include/opae/plugin/adapter.h

RPM build errors:
    File listed twice: /usr/lib/.build-id/b8/e0e176bef6cc46ac0d9d542613d3eef9108e90
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/PKG-INFO
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/SOURCES.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/dependency_links.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/entry_points.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/requires.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae.fpga-2.1.0-py3.8.egg-info/top_level.txt
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__init__.py
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__pycache__
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__pycache__/__init__.cpython-38.opt-1.pyc
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/__pycache__/__init__.cpython-38.pyc
    File listed twice: /usr/lib64/python3.8/site-packages/opae/fpga/_opae.cpython-38-x86_64-linux-gnu.so
    Installed (but unpackaged) file(s) found:
   /usr/include/opae/plugin/adapter.h
first rpmbuild failed.. exiting

tswhison, 2023-06-02:
Fedora32 is legacy.  Code is packaging correctly with Fedora35, Fedora37.

closed 2023-06-02 by tswhison