OFS/opae-sdk #2613: can it run on arm /aarch64/ armv8l?

AndreV84, 2022-05-28:
it won't build on arm64

tswhison, 2022-06-03:
Hello, it seems that __builtin_cpu_init() is not available on your platform. It seems we should add a preprocessor check for __arm__, perhaps?

AndreV84, 2022-06-03:
@tswhison 
how can it be added?

tswhison, 2022-06-03:
Could you try this branch?  #2616

AndreV84, 2022-06-03:
@tswhison 
seems like in https://github.com/google/or-tools/issues/820#issuecomment-414224256

tswhison, 2022-06-03:
Ah yes.  Thank you.  I don't have an ARM test bench.  We wouldn't be able to compile those instructions for non-x86 I think.  I have updated the branch.

AndreV84, 2022-06-04:
maybe we can somehow patch the code to pass ther err, e.g. as in https://github.com/google/snappy/blob/main/snappy-stubs-internal.h#L70 ?

tswhison, 2022-06-06:
Right now, the code will return FPGA_NOT_SUPPORTED for the 512-bit calls on arm.  Is that what you mean?

tswhison, 2022-06-06:
I thought I had fixed that one.  Could you try updating your branch for #2616?

AndreV84, 2022-06-06:
@tswhison Thank you for the update!
Seems I can build it on my aarch64 device
is there a simple test/ to try the board with it?
https://www.intel.com/content/dam/www/central-libraries/us/en/documents/intel-pac-with-arria-10-gx-product-brief.pdf

AndreV84, 2022-06-06:
could you extend what needs to be done after executing the step below, please?
`git clone https://github.com/OPAE/linux-dfl`

tswhison, 2022-06-06:
You'll need to check with the folks over at linux-dfl.  I'm not sure whether an ARM driver exists, but they would know for certain.

AndreV84, 2022-06-09:
@tswhison 
could you advise also how to build these on x86_64 that got installed intel dev stack environment but doesn't seem to have built sample binaries for opae-sdk
```
to build these guys?

usr/src/opae/samples$ ls
hello_events.c hello_fpga.c
```

tswhison, 2022-06-09:
Please find build instructions for hello_fpga here: https://opae.github.io/0.13.0/docs/fpga_api/quick_start/readme.html#building-a-sample-application

AndreV84, 2022-06-09:
found 
https://www.intel.com/content/www/us/en/docs/programmable/683633/1-2-1/running-the-hello-fpga-example.html
thank you very much

AndreV84, 2022-06-10:
``` lspci | grep 09c4
07:00.0 Processing accelerators: Intel Corporation Device 09c4 (rev ff)
```

AndreV84, 2022-06-11:
on 20.04 ubuntu the intel dev stack script won't run
will opae-sdk/ driver build from source?

AndreV84, 2022-06-11:
@tswhison trying to build linux-dfl according to opae-sdk tutorial
on x86_64 20.04 ubuntu
`]  fs/jffs2/scan.o
  CC [M]  net/mac80211/vht.o
  CC [M]  drivers/leds/leds-sgm3140.o
  CC      drivers/acpi/acpica/exfield.o
gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
  CC [M]  drivers/video/backlight/88pm860x_bl.o
`

tswhison, 2022-06-13:
Could you please raise this at linux-dfl?  They should be able to help with driver issues.

nathanborak, 2024-10-14:
Closing due to inactivity/age, and because the original question was answered. If you still require assistance, please re-open the issue.

closed 2024-10-14 by nathanborak