OFS/opae-sdk #2638: Multiple clock domains in same AFU

charleshong3, 2022-06-27:
To my knowledge it is currently not possible to have multiple clocks in one AFU. Is there any chance this feature will be added in future releases?

michael-adler, 2022-07-08:
There are a few problems you'd have to manage with clock gating:
* AFUs must continue to respond to MMIO traffic on the PCIe bus.
* All clock management must be outside the PR region.

You could run the majority of a design on uClk and change its frequency to something low when idle, which might be similar to gating. Nothing in the FIM depends on uClk, so you are free to set it as low as you wish.

Once OFS source for the FIM is released it would be possible to define more clocks and manage them as needed. My hunch is that the power consumption would be similar to changing uClk.

charleshong3, 2022-07-08:
Thanks for the reply! We are clock gating for functional reasons, not power, so limiting uClk unfortunately doesn't work for us. This is related to the effort to run [FireSim](https://fires.im/) locally on Intel FPGAs, which I suspect you've probably heard about previously. In FireSim, clock gating is required to maintain cycle accuracy guarantees when simulating operations like DRAM accesses.

Will the OFS source release allow for this functionality? And is there a timeline on this release?

Thanks again.

charleshong3, 2023-02-17:
Hi, just want to ask if the status of this issue has changed at all?

michael-adler, 2023-02-17:
A reference implementation of OFS RTL for the d5005 board is public now, spread between https://github.com/OFS/ofs-d5005 and https://github.com/OFS/ofs-fim-common. An Agilex reference design is scheduled for April.

PR doesn't support clock gating, so you'd have to build your entire design in a flat build. That is not the default in the public release, but it is possible.

closed 2023-12-23 by michael-adler