OFS/opae-sdk #2841: PACSign fails on RHEL9 because of OpenSSL version 3.0.1

rchriste1, 2023-02-10:
Running PACSign on RHEL 9 and 9.1 (RockyLinux9) fails with:
2023-02-10 08:55:25,241 - PACSign.log - DEBUG - OpenSSL version "OpenSSL 3.0.1 14 Dec 2021" fails to match "OpenSSL 3.0.2 15 Mar 2022"
Error: Module: openssl.py, Function: _find_openssl_so, Line: 330
       Failed to find crypto library (libcrypto.so).

Updating package openssl-devel does not provide OpenSSL 3.0.2, which is the only one PACSign accepts of 3.x.x.

If I add ('libcrypto.so.3', 'OpenSSL 3.0.1 14 Dec 2021') to the list of versions, it works. However, I am wondering if it is necessary that PACSign requires a specific version among 3.x.x. New versions are coming frequently. I think 3.0.2 was added to PACSign because it is the one in Ubuntu 22.04.

tswhison, 2023-02-10:
Hello rchriste1, thanks for pointing this out.  PACSign does require a particular interface from libcrypto.so in order to work, but it seems to work well with version >= 1.1.1.  Would you be able to provide the patch you got to work in your environment?

tswhison, 2023-02-15:
Fixed by #2842.

closed 2023-02-15 by tswhison