OFS/opae-sdk #2912: Unable to build RPM due to missing pybind packages

wespiard, 2023-04-06:
I'm trying to build the OPAE RPMs on a system running Rocky Linux 8.7 (x86_64). 

When I run the `./create fedora` script to build the RPMs, it gets stuck here, because it cannot find `python3-pybind11` or `pybind11-devel`: https://github.com/OFS/opae-sdk/blob/8a35db74e40ce9b2a89cea3eba764dda8a22927b/packaging/opae/rpm/create#L89-L102

These pybind packages do not appear to be available via `dnf`. I noticed that the [instructions for Ubuntu](https://opae.github.io/2.3.0/docs/install_guide/installation_guide.html#ubuntu-20-04) install pybind11 via pip, instead of a package manager.

tswhison, 2023-04-06:
Hi @wespiard, sorry that you're having trouble.  Have you tried the instructions here, specifically setting up powertools and epel-release?  https://ofs.github.io/sw/install_guide/installation_guide/#rocky-linux-85

I think that this will enable you.  Give it a shot and let me know how it works out.

wespiard, 2023-04-06:
Hey @tswhison, 

Thanks for the quick reply! Those are pretty much the exact commands I ran to start. Below is the output I get when I re-run them.

The first issue arose when running `dnf install`: 
`Error: Unable to find a match: python3-pybind11 pybind11-devel`

```
❯ sudo dnf install -y 'dnf-command(config-manager)'
Last metadata expiration check: 2:52:09 ago on Thu 06 Apr 2023 02:42:44 PM UTC.
Package dnf-plugins-core-4.0.21-14.1.el8.noarch is already installed.
Dependencies resolved.
Nothing to do.
Complete!

❯ sudo dnf config-manager --set-enabled powertools

❯ sudo dnf install -y epel-release
Rocky Linux 8 - PowerTools                                                 9.6 kB/s | 4.8 kB     00:00
Package epel-release-8-18.el8.noarch is already installed.
Dependencies resolved.
Nothing to do.
Complete!

❯ sudo dnf check-update
Last metadata expiration check: 0:00:09 ago on Thu 06 Apr 2023 05:35:10 PM UTC.

❯ sudo dnf upgrade -y
Last metadata expiration check: 0:00:19 ago on Thu 06 Apr 2023 05:35:10 PM UTC.
Dependencies resolved.
Nothing to do.
Complete!

❯ sudo dnf install -y python3 python3-pip python3-devel python3-jsonschema python3-pyyaml python3-pybind11 git gcc gcc-c++ make cmake libuuid-devel json-c-devel hwloc-devel tbb-devel cli11-devel spdlog-devel libedit-devel systemd-devel rpm-build rpmdevtools pybind11-devel yaml-cpp-devel libudev-devel linuxptp
Last metadata expiration check: 0:00:32 ago on Thu 06 Apr 2023 05:35:10 PM UTC.
Package python36-3.6.8-38.module+el8.5.0+671+195e4563.x86_64 is already installed.
Package python3-pip-9.0.3-22.el8.rocky.0.noarch is already installed.
Package python36-devel-3.6.8-38.module+el8.5.0+671+195e4563.x86_64 is already installed.
Package python3-jsonschema-2.6.0-4.el8.noarch is already installed.
Package python3-pyyaml-3.12-12.el8.x86_64 is already installed.
No match for argument: python3-pybind11
Package git-2.31.1-3.el8_7.x86_64 is already installed.
Package gcc-8.5.0-16.el8_7.x86_64 is already installed.
Package gcc-c++-8.5.0-16.el8_7.x86_64 is already installed.
Package make-1:4.2.1-11.el8.x86_64 is already installed.
Package cmake-3.20.2-4.el8.x86_64 is already installed.
Package libuuid-devel-2.32.1-39.el8_7.x86_64 is already installed.
Package json-c-devel-0.13.1-3.el8.x86_64 is already installed.
Package hwloc-devel-2.2.0-3.el8.x86_64 is already installed.
Package tbb-devel-2018.2-9.el8.x86_64 is already installed.
Package cli11-devel-1.8.0-2.el8.x86_64 is already installed.
Package spdlog-devel-1.5.0-2.el8.x86_64 is already installed.
Package libedit-devel-3.1-23.20170329cvs.el8.x86_64 is already installed.
Package systemd-devel-239-68.el8_7.4.x86_64 is already installed.
Package rpm-build-4.14.3-24.el8_7.x86_64 is already installed.
Package rpmdevtools-8.10-8.el8.noarch is already installed.
All matches were filtered out by modular filtering for argument: pybind11-devel
Package yaml-cpp-devel-0.6.2-3.el8.x86_64 is already installed.
Package systemd-devel-239-68.el8_7.4.x86_64 is already installed.
Package linuxptp-3.1.1-3.el8.x86_64 is already installed.
Error: Unable to find a match: python3-pybind11 pybind11-devel
```

wespiard, 2023-04-06:
That is odd. I tried doing `dnf search pybind` and some other things, but neither of those were showing up even though I have the powertools repo installed. 

I installed those two RPMs manually, like this:
```bash
wget https://download.rockylinux.org/pub/rocky/8/PowerTools/x86_64/os/Packages/p/python39-pybind11-2.7.1-1.module%2Bel8.6.0%2B795%2Bde4edbcc.x86_64.rpm
wget https://download.rockylinux.org/pub/rocky/8/PowerTools/x86_64/os/Packages/p/python39-pybind11-devel-2.7.1-1.module%2Bel8.6.0%2B795%2Bde4edbcc.x86_64.rpm
sudo dnf in -y *.rpm
```

Then, updated the `create` script and `opae.spec.fedora` file to look for them instead of the old names. **This got me past the error checks in the `create` script.** 

Unfortunately, though, it failed late into the actual build, because it's using `/usr/libexec/platform-python` which is version `3.6.8` and the `setup.py` script is trying to import `find_namespace_packages` from `setuptools`:

```bash
+ pushd /home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/opae.io
+ /usr/libexec/platform-python setup.py install --single-version-externally-managed --root=/home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILDROOT/opae-2.3.0-1.el8.x86_64
Traceback (most recent call last):
  File "setup.py", line 27, in <module>
    from setuptools import find_namespace_packages
ImportError: cannot import name 'find_namespace_packages'
error: Bad exit status from /var/tmp/rpm-tmp.TmvUiZ (%install)
    Bad exit status from /var/tmp/rpm-tmp.TmvUiZ (%install)
```

I assume `find_namespace_packages` is not in the version of `setuptools` that comes with `Python 3.6.8`?

I tried importing it as a test using `platform-python` and `python39` to illustrate:

```bash
$ python3 -c "from setuptools import find_namespace_packages"
$ /usr/libexec/platform-python -c "from setuptools import find_namespace_packages"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'find_namespace_packages'
```

tswhison, 2023-04-06:
Oh, good progress.  For the setuptools issue, you can update setuptools with pip3:
```bash
$ sudo pip3 uninstall setuptools
$ sudo pip3 install --upgrade setuptools --prefix=/usr
```

wespiard, 2023-04-06:
After running those commands, and tryng `./create fedora` again, I get the following error:

```bash
error in pyopaeuio setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after name and no valid version specifier)
    pybind11>=@PYOPAE_PYBIND11_VERSION@
            ^
make[2]: *** [libraries/pyopaeuio/CMakeFiles/pyopaeuio-build.dir/build.make:78: libraries/pyopaeuio/timestamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:1619: libraries/pyopaeuio/CMakeFiles/pyopaeuio-build.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:159: all] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.AUF8Jj (%build)
    Bad exit status from /var/tmp/rpm-tmp.AUF8Jj (%build)
```

tswhison, 2023-04-06:
Hmm, that's a variable reference that didn't seem to get substituted: @PYOPAE_PYBIND11_VERSION@
Can you run the clean script in the directory and then try create once more?
If that doesn't fix it, try replacing that variable reference manually with 2.10.2

wespiard, 2023-04-06:
Running it after the clean script yielded the same error. 

Re-running the create script after manually changing the variable reference to 2.10.2 overwrote the file, putting back the variable reference.

I even tried running just the `rpmbuild` call from the create script, but still gets overwritten. I guess because it's copying the opae*.tar.gz archive from the SOURCES directory. 

So then I extracted this directory, changed the two setup files that have that variable, re-tarzipped it, and then re-ran the rpm-build command and got this error again:
```bash
+ pushd /home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/opae.io
~/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/opae.io ~/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1
+ /usr/libexec/platform-python setup.py install --single-version-externally-managed --root=/home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILDROOT/opae-2.3.0-1.el8.x86_64
Traceback (most recent call last):
  File "setup.py", line 27, in <module>
    from setuptools import find_namespace_packages
ImportError: cannot import name 'find_namespace_packages'
```

So I changed the version that `python3` was point to from 3.9 to 3.6, and re-updated setuptools, but got the same error again:

```bash
cd /home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/fpgadiag && /usr/bin/python3 setup.py build_ext -I /home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/include:/home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/external/pybind11/include -L /home/wespiard/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/lib
Traceback (most recent call last):
  File "setup.py", line 26, in <module>
    from setuptools import setup, find_namespace_packages
ImportError: cannot import name 'find_namespace_packages'
```

tswhison, 2023-04-06:
>>Re-running the create script after manually changing the variable reference to 2.10.2 overwrote the file, putting back the variable reference.

Did you change the files in your working tree or inside the tarball directory that create is building?  If you update the files in your source tree, they shouldn't get reset by the create script.  It simply copies the files into the tarball.

I see two references to @PYOPAE_PYBIND11_VERSION@ in the source here:
opae-sdk/libraries/pyopae/setup.py
opae-sdk/libraries/pyopaeuio/setup.py

You could try substituting both of those with 2.10.2, then clean, and create fedora.  The changes should be copied into the tarball dir.

Sorry for so much trouble.  This is currently working in Fedora for some reason.  I'm not sure why we're having so much trouble with RockyLinux, but we'll get to the bottom of it.

wespiard, 2023-04-06:
> I see two references to @PYOPAE_PYBIND11_VERSION@ in the source here:
> opae-sdk/libraries/pyopae/setup.py
> opae-sdk/libraries/pyopaeuio/setup.py

Those were the files I changed. I misinterpreted how the create script worked. I guess I assumed it was downloading the tarball from GitHub or something, not archiving from the local source tree. 

So, I'm getting this same error after cleaning, changing the two variable instances to 2.10.2 manually, and re-running create:
```bash
cd /home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/fpgadiag && /usr/bin/python3 setup.py build_ext -I /home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/include:/home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/external/pybind11/include -L /home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/lib
Traceback (most recent call last):
  File "setup.py", line 26, in <module>
    from setuptools import setup, find_namespace_packages
ImportError: cannot import name 'find_namespace_packages'
make[2]: *** [binaries/fpgadiag/CMakeFiles/opae.diag-build.dir/build.make:89: binaries/fpgadiag/timestamp] Error 1
make[2]: Leaving directory '/home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1'
make[1]: *** [CMakeFiles/Makefile2:2458: binaries/fpgadiag/CMakeFiles/opae.diag-build.dir/all] Error 2
```

Not sure if it helps or not, but but I get the following warnings/errors when I change `/usr/bin/python3` to point to python39 instead of python36:

```bash
+ echo 'Installing for RHEL <= 8'
+ /usr/bin/make install DESTDIR=/home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILDROOT/opae-2.3.0-1.el8.x86_64 'INSTALL=/usr/bin/install -p'
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/dist.py:561: UserWarning: The version specified ('1.4.3+*') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/_normalization.py:93: SetuptoolsDeprecationWarning: Invalid version: '1.4.3+*'.
        !!

        ###################
        # Invalid version #
        ###################
        '1.4.3+*' is not valid according to PEP 440.

        Please make sure specify a valid version for your package.
        Also note that future releases of setuptools may halt the build process
        if an invalid version is given.

!!

  warnings.warn(cleandoc(msg), SetuptoolsDeprecationWarning)
Traceback (most recent call last):
  File "/home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/python/opae.admin/setup.py", line 29, in <module>
    setup(
  File "/usr/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/usr/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/usr/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/usr/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.9/site-packages/setuptools/command/install.py", line 68, in run
    return orig.install.run(self)
  File "/usr/lib/python3.9/site-packages/setuptools/_distutils/command/install.py", line 708, in run
[... 21 lines cut ...]
  warnings.warn(msg)
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
+ prev=/home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1
+ pushd /home/wespiard/git/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/opae.io
[... cut ...]
The main difference I see is that for the first error it's calling `/usr/bin/python3 setup.py build_ext ...` and in the second, it's calling `/usr/libexec/platform-python setup.py install ...`. 

Another interesting note is that I can import `find_namespace_packages` in any of the versions of the interpreter that I try: 

```bash 
❯ /usr/libexec/platform-python -c "from setuptools import find_namespace_packages"
❯ /usr/bin/python3.6 -c "from setuptools import find_namespace_packages"
❯ /usr/bin/python3.9 -c "from setuptools import find_namespace_packages"
```

wespiard, 2023-04-07:
Got it working! 

I ended up starting with a fresh install of Rocky 8.7 (same as above). I think I must have had some weird misconfiguration with my Python installations.

Here are the steps I took to get it working, in case anyone has a similar issue by chance:

```bash
sudo dnf install -y python3 python3-pip python3-devel python3-jsonschema python3-pyyaml \
    git gcc gcc-c++ make cmake libuuid-devel json-c-devel hwloc-devel tbb-devel cli11-devel \
    spdlog-devel libedit-devel systemd-devel rpm-build rpmdevtools yaml-cpp-devel libudev-devel linuxptp

# Upgrade `setuptools` and install `pybind11`
pip3 install --user --upgrade setuptools  pybind11

# Change the RPM build requirements in the `packaging/opae/rpm/create` and `opae.spec.fedora` files.
cd <opae_sdk_repo>
sed 's/python3-pybind11/python39-pybind11/g' opae.spec.fedora -i
sed 's/pybind11-devel/python39-pybind11-devel/g' opae.spec.fedora -i
sed 's/python3-pybind11/python39-pybind11/g' packaging/opae/rpm/create -i
sed 's/pybind11-devel/python39-pybind11-devel/g' packaging/opae/rpm/create -i

cd <opae_sdk_repo>/packaging/opae/rpm
./create fedora
```

I am now no longer sure what the issue with the `@..PYBIND..@` variable references. May be related to the way I'm installing pybind11 now?

wespiard, 2023-04-07:
I also just tried removing the `pybind11` requirements from the `create` and `opae.spec.fedora` files, and exclusively installing it with pip.

Not sure if this is proper in general, and makes the build steps less OS-dependent, or if it's just a fluke with my setup.

--- 

Start by installing required packages and cloning the repo.

```bash
sudo dnf install -y python3-devel python3-jsonschema python3-pyyaml git gcc gcc-c++ make cmake json-c-devel libuuid-devel rpm-build systemd rpmdevtools tbb-devel hwloc-devel cli11-devel spdlog-devel libedit-devel systemd-devel
pip3 install --user --upgrade setuptools pybind11
git clone https://github.com/OPAE/opae-sdk.git
```

Then, from inside the repo, delete the following lines:
https://github.com/OFS/opae-sdk/blob/2065ed57e829fd7bc947e5f672925e0a07db9445/packaging/opae/rpm/create#L93
https://github.com/OFS/opae-sdk/blob/2065ed57e829fd7bc947e5f672925e0a07db9445/packaging/opae/rpm/create#L96
https://github.com/OFS/opae-sdk/blob/2065ed57e829fd7bc947e5f672925e0a07db9445/opae.spec.fedora#L37
https://github.com/OFS/opae-sdk/blob/2065ed57e829fd7bc947e5f672925e0a07db9445/opae.spec.fedora#L42

Then, run the `create` script to generate the RPMs.

```bash
cd opae-sdk/packaging/opae/rpm
./create fedora
```

tswhison, 2023-04-07:
I'm glad you got it working.  I was going to suggest that next thing to try as looking at your Python setup. :)
It's unfortunate that these package names are changing on certain distros.  It makes developing for the generic case very difficult.
The @PYOPAE_PYBIND11_VERSION@ issue has been addressed on master. We no longer rely on this variable.
Also, we will soon be moving all of our python projects to pyproject.toml, per [PEP 517](https://peps.python.org/pep-0517/).

wespiard, 2023-04-07:
Yeah, it would be nice if they stayed consistent. I wonder if there are better ways of verifying a package is installed like `ldconfig -p` for libraries, etc., or matching other patterns instead of the full package name.

Anyways, thanks for all of the tips yesterday!

closed 2023-04-07 by wespiard