OFS/opae-sdk #2946: ollect2: error: ld returned 1 exit status make[2]: *** [binaries/opae.io/CMakeFiles/opae.io.dir/build.make:104: bin/opae.io] Error 1 make[2]: Leaving directory '/root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1' make[1]: *** [CMakeFiles/Makefile2:2626: binaries/opae.io/CMakeFiles/opae.io.dir/all] Error 2

208yunyao, 2023-05-16:
cd /root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/opae.io && /usr/bin/cmake -E cmake_link_script CMakeFiles/opae.io.dir/link.txt --verbose=1
/usr/bin/c++ -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -Wformat -Wformat-security -fPIC -D_FORTIFY_SOURCE=2 -fstack-protector-all -std=c++14 -std=gnu++11 -g -O2 -Wall -Wextra -Werror -pthread -Wl,-z,relro  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -pie CMakeFiles/opae.io.dir/main.cpp.o -o ../../bin/opae.io  -Wl,-rpath,/root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/lib: -ldl -lutil /usr/lib64/libedit.so ../../lib/libopaevfio.so.2.3.0 /usr/local/lib/libpython3.10.a -lpthread ../../lib/libopaemem.so.2.3.0
/usr/bin/ld: /usr/local/lib/libpython3.10.a(abstract.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(boolobject.o): relocation R_X86_64_32S against symbol `_Py_TrueStruct' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(bytearrayobject.o): relocation R_X86_64_32S against symbol `PyByteArray_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(bytesobject.o): relocation R_X86_64_32S against symbol `PyBytes_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(call.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(capsule.o): relocation R_X86_64_32S against symbol `PyCapsule_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(classobject.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(complexobject.o): relocation R_X86_64_32S against symbol `PyComplex_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(descrobject.o): relocation R_X86_64_32 against symbol `PyClassMethodDescr_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(exceptions.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(genericaliasobject.o): relocation R_X86_64_32 against symbol `Py_GenericAliasType' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(genobject.o): relocation R_X86_64_32 against symbol `PyCoro_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(floatobject.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(frameobject.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(iterobject.o): relocation R_X86_64_32 against symbol `PyCallIter_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(listobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(longobject.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(dictobject.o): relocation R_X86_64_32 against symbol `PyDictValues_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(memoryobject.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(methodobject.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(moduleobject.o): relocation R_X86_64_32 against symbol `PyModule_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(object.o): relocation R_X86_64_32 against symbol `PyBaseObject_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(obmalloc.o): relocation R_X86_64_32S against `.text.hot' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(picklebufobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(rangeobject.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(setobject.o): relocation R_X86_64_32S against symbol `PySet_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(sliceobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(structseq.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(tupleobject.o): relocation R_X86_64_32 against symbol `PyTuple_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(typeobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(unicodeobject.o): relocation R_X86_64_32 against symbol `PyUnicode_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(unicodectype.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(unionobject.o): relocation R_X86_64_32S against symbol `_Py_NoneStruct' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(weakrefobject.o): relocation R_X86_64_32 against `.bss' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(_warnings.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(ceval.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(codecs.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(compile.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(errors.o): relocation R_X86_64_32 against `.data' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(future.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(getargs.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(import.o): relocation R_X86_64_32S against symbol `PyModule_Type' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libpython3.10.a(importdl.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
[... cut ...]
make[2]: Entering directory '/root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1'
[ 95%] Generating timestamp
cd /root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/binaries/fpgadiag && /usr/local/bin/python3 setup.py build_ext -I /root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/include:/include -L /root/opae-sdk/packaging/opae/rpm/rpmbuild/BUILD/opae-2.3.0-1/lib
running build_ext
building 'opae.diag.eth_group' extension
creating build
creating build/temp.linux-x86_64-cpython-310

tswhison, 2023-05-16:
Sorry that you're having trouble.  May I ask, which distribution are you compiling in?  I may need to reproduce your environment.

tswhison, 2023-06-02:
No traffic in two weeks.. Closing

closed 2023-06-02 by tswhison