OFS/opae-sdk #2996: error in bel_power_regulator_info of N6000 Board

umairsiddiqui-digitek, 2023-08-07:
found error in bel_power_regulator_info of N6000 Board: 

https://github.com/OFS/opae-sdk/blob/4cb9d59102415b6393b784502c7fd1d37366066b/libraries/libboard/board_n6000/board_event_log.c#L163C1-L176C3

**BEL_PWR_REG_IR38062_VIN** & **BEL_PWR_REG_IR38062_TEMP** have incorrect labels and units
similarly 

**BEL_PWR_REG_IR38063_VIN** & **BEL_PWR_REG_IR38063_TEMP**
and 
**BEL_PWR_REG_ISL68220_VIN** & **BEL_PWR_REG_ISL68220_TEMP**
are also incorrect 

it should be 

```
static struct bel_sensor_info bel_power_regulator_info[] = {
	[BEL_PWR_REG_IR38062_VOUT]  = { .label = "IR38062 Voltage",      .unit = "mV", .resolution = 1 },
	[BEL_PWR_REG_IR38062_IOUT]  = { .label = "IR38062 Current",      .unit = "mA", .resolution = 1 },
	[BEL_PWR_REG_IR38062_VIN]   = { .label = "IR38062 Input",        .unit = "mV", .resolution = 1 },
	[BEL_PWR_REG_IR38062_TEMP]  = { .label = "IR38062 Temperature",  .unit = "°C", .resolution = 1 },
	[BEL_PWR_REG_IR38063_VOUT]  = { .label = "IR38063 Voltage",      .unit = "mV", .resolution = 1 },
	[BEL_PWR_REG_IR38063_IOUT]  = { .label = "IR38063 Current",      .unit = "mA", .resolution = 1 },
	[BEL_PWR_REG_IR38063_VIN]   = { .label = "IR38063 Input",        .unit = "mV", .resolution = 1 },
	[BEL_PWR_REG_IR38063_TEMP]  = { .label = "IR38063 Temperature",  .unit = "°C", .resolution = 1 },
	[BEL_PWR_REG_ISL68220_VOUT] = { .label = "ISL68220 Voltage",     .unit = "mV", .resolution = 1 },
	[BEL_PWR_REG_ISL68220_IOUT] = { .label = "ISL68220 Current",     .unit = "mA", .resolution = 1 },
	[BEL_PWR_REG_ISL68220_VIN]  = { .label = "ISL68220 Input",       .unit = "mV", .resolution = 1 },
	[BEL_PWR_REG_ISL68220_TEMP] = { .label = "ISL68220 Temperature", .unit = "°C", .resolution = 1 }
};

```

aditikum, 2023-08-08:
Hi @umairsiddiqui-digitek, thank you for notifying us of this issue. We'll review it and assign someone to address it

anandaravuri, 2023-08-08:
Fixed in PR https://github.com/OFS/opae-sdk/pull/2997

closed 2023-08-11 by aditikum