OFS/opae-sdk #3024: (security vulnerability) use of python pickle module in opae.io python scripts

umairsiddiqui-digitek, 2023-10-11:
i want to point out the use of python pickle module in python script of opae.io. 
https://github.com/OFS/opae-sdk/blob/4cb9d59102415b6393b784502c7fd1d37366066b/binaries/opae.io/opae/io/utils.py#L54C4-L54C4

 As opae.io binary run as root, there is a (possible) security risk of arbitrary code execution during unpickling

https://docs.python.org/3/library/pickle.html

> Warning The pickle module is not secure. Only unpickle data you trust.
> It is possible to construct malicious pickle data which will execute arbitrary code during unpickling. Never unpickle data that could have come from an untrusted source, or that could have been tampered with.
> 
> Safer serialization formats such as [json](https://docs.python.org/3/library/json.html#module-json) may be more appropriate if you are processing untrusted data. See [Comparison with json](https://docs.python.org/3/library/pickle.html#comparison-with-json).
>

tswhison, 2023-10-11:
Thank you for pointing this out.  The pickle file itself is also protected by root permissions.
```bash
-rw-r--r--. 1 root root 66 Oct 11 09:37 /var/lib/opae/opae.io.pickle
```
If someone is able to inject malicious data into the pickle file, that implies that they already have escalated privileges (or else, how could they have written to the file?)  From that perspective, do you think there is still some issue?

I will investigate converting it to JSON format.

tswhison, 2023-10-17:
Resolved by #3028.

closed 2023-10-17 by tswhison