OFS/opae-sdk #3032: N6000/N6001 sudo fpgainfo errors all returns error code 70

rchriste1, 2023-10-25:
Running
```shell
sudo fpgainfo errors all
```
With N6000 or N6001 card gives the correct output of FME and PORT errors, but return error code 70. When running with either argument "fme" or "port" instead of "all"  the error code is zero as expected.

It seems the problem occurs due to the addition of UIO interfaces for N6000/N6001 in commit 029ef77023, that can be seen with
```shell
sudo fpgainfo port
``` 
These devices are also found in the "errors all" command, but "fpgaGetProperties(tokens[i], &props);" in errors-command() returns error 4, which result in the returned error code.

This does not happen in the "sudo fpgainfo errors port" case, because a FPGA_IFC_DFL filter is added. If the same filter is added in the "VERB_ALL" case as shown here, the UIO devices are not included so the problem seems to be solved with adding these 4 lines.

```c
		case VERB_ALL:
			res = fpgaPropertiesSetInterface(*filter,
				FPGA_IFC_DFL);
			ON_FPGAINFO_ERR_GOTO(
				res, out, "setting type to FPGA_IFC_DFL");

``` 
This also works for N501x cards, but I do not know if it leads to problems for other cards.

rchriste1, 2023-11-08:
Hi @anandaravuri , do you need more information? should I create a pull request with the above proposed solution?

nathanborak, 2024-10-14:
Hey @anandaravuri, @rchriste1, can we consider this issue solved, there haven't been any updates in nearly a year.

closed 2024-10-28 by nathanborak